cucapra/dahlia #16: Index type addition issues

sa2257, 2018-09-07:
Following program passes

`
func mmul(a: float[9 bank(3)], b: float[9 bank(3)], c: float[9 bank(3)]) {

  for (let i = 0..3) unroll 1 {
    for (let j = 0..3) unroll 3 {
      c[i] := a[i*3+j] + b[i*3+j];
    }
  }

}
`

But following fails with error `[Type error] can't apply operator '+' to idx<0 .. 1, -4611686018427387904 .. 4611686018427387903> and idx<0 .. 3, 0 .. 1>`

`
func mmul(a: float[9 bank(3)], b: float[9 bank(3)], c: float[9 bank(3)]) {

  for (let i = 0..3) unroll 1 {
    for (let j = 0..3) unroll 3 {
      c[i*3+j] := a[i*3+j] + b[i*3+j];
    }
  }

}
`
Shouldn't both give the same error?

sampsyo, 2018-09-07:
Yo, @sa2257: for future reference, you can use a code block (```) to preserve line wrapping in pastes like these.

sa2257, 2018-09-07:
Oh yes!! Thanks!

rachitnigam, 2018-09-17:
Reassignment type checking was implemented incorrectly IMO. Both examples now fail on master. We need to formalize addition on index types before either example will work.

rachitnigam, 2019-01-31:
Views subsume this

closed 2019-01-31 by rachitnigam