cucapra/dahlia #23: Benchmarking and evaluation

rachitnigam, 2018-09-11:
Since the long term goal of seashell is to demonstrate its superiority over current HLS implementations, we will need to port and test existing HLS benchmarks (even make our own?).

@sampsyo has suggested looking at the following suites:
- MachSuite https://breagen.github.io/MachSuite/
- Rosetta https://github.com/cornell-zhang/rosetta
- Spatial’s apps https://github.com/stanford-ppl/spatial-apps

We should start collecting the HLS kernels from these and put them in the folder. The type checker people will stare at them really hard and come up with language features and turn them into test cases.

rachitnigam, 2019-01-31:
This is to keep track of the status of MachSuites apps in Seashell. Subsequent posts will track each app.

- [ ] gemm ncubed
- [ ] stencil2d

sa2257, 2019-01-31:
I believe this issue is to keep track of Benchmarks we generate. #45 is to handle errors and gaps found when running MachSuite apps.

rachitnigam, 2019-01-31:
No, this is the tracker for tracking benchmarking gaps. We keep track of benchmarks we generate but putting them in a folder

rachitnigam, 2019-02-18:
Proposals
----
**Usability evaluation**:
Count the number of `flex` view instances we had to use to make the benchmarks work. Since `flex` are considered bad (they need to create expensive and arbitrary hardware for accesses). If we have expressive constructs, we won't need a lot of `flex`.

**Expressiveness evaluation**:
We might find programs that are correct according to our type system but rejected by HLS (see https://github.com/cucapra/seashell/issues/63#issuecomment-464841178). This is a win for us. We can argue for type-directed design by showing that our type system can prove a larger set of accesses safe than the random static analysis used by HLS compilers.

rachitnigam, 2019-02-26:
CHStone: https://github.com/davestevens/LE1/tree/master/benchmarks/legup/chstone

Used in the original LegUp HLS paper

rachitnigam, 2019-03-11:
PolyBench: http://web.cse.ohio-state.edu/~pouchet.2/software/polybench/

sa2257, 2019-03-15:
Since 'using MachSuite as a performance benchmark' came up again today, I'll just recap what I thought our conclusion was about a month back when we discussed this.

I think we agreed MachSuite is mainly a code coverage benchmark for HLS. But we prioritize covering it as a benchmark since it'll be useful to have a full benchmark and due to lack of reasonable benchmarks suites out there. 

Meanwhile, I'm using this [tracker](https://github.com/cucapra/epoisses/issues/3) to evaluate other kernels in benchmarks suites to come up with a reasonable set of apps to evaluate Seashell with Vivado HLS.

sa2257, 2019-03-15:
CHStone seems like another useful suite. I didn't add it to HLS list yet because I'm not sure what there goal with this suite. It has some really cool apps like a MIPS processor, JPEG compression and MPEG-2 compression. But I'm interested whether this is just an attempt to extend expressivity of LegUP using reference C versions of these, or an actual attempt to come up with optimized designs. More domain knowledge is needed.

rachitnigam, 2019-03-15:
I mostly agree with Sachille on machsuite not being a perf benchmark suite. However, there is an orthogonal point I’ve been trying to explain about the *integrity* of benchmarking. If we port a benchmark from *any* suite, we need to be very precise about which pragmas we keep.

Sachille said that the commented out unrolls should not be discounted since they represent the design exploration that machsuite people did for their setup. 

This is a good point. Regardless of which purpose machsuite (or any suite) serves, we have to choose a precise set of pragmas and describe it in the paper so people can reproduce our numbers.

For example, we might either decide to only use the uncommented pragmas *or* choose to use the set of pragmas that makes the c++ machsuite benchmarks performance the best it possibly can.

For now, I recommend choosing the former.

sa2257, 2019-03-15:
Well, let's certainly go with the former for now, as tuning the HLS code might open the Pandora's box on how good this code is. But, in order to do so, one critical thing we need to add is pipelining. I do not think we can reasonably compete without pipelining. It can give several orders of speedup. (Hence why I have been selectively adding pragmas)

rachitnigam, 2019-03-15:
Yeah, that sounds reasonable. Regardless of the state of Fuse, we should use any uncommented pragmas and add features to Fuse as we go on.

sampsyo, 2019-03-16:
Y'all made some good points above in this thread!

> Regardless of which purpose machsuite (or any suite) serves, we have to choose a precise set of pragmas and describe it in the paper so people can reproduce our numbers.

Yep, for fairness, clarity of explanation, and comparability, we need to have a succinct way to describe the annotations we used. An easy way to do that would be to use the annotations as-is (either with or without the commented-out ones) from MachSuite; doing our own tuning or hand-picking pragmas makes this harder to explain (and opens the door the the *appearance* of selecting pragmas that make us look good!).

> tuning the HLS code might open the Pandora's box on how good this code is

Indeed; really working hard to tune the HLS code is a Pandora's box. Doing so has the advantage of giving a truly competitive baseline, but it would (a) take a lot of time and (b) still not be terribly easy to describe—readers would have to "take our word for it" that we actually found an optimal-ish design point.

rachitnigam, 2019-07-02:
[Rodania](https://rodinia.cs.virginia.edu/doku.php): Heterogeneous computing using openMP, openCL, and CUDA. Used as [example](https://github.com/Xilinx/SDAccel_Examples) in the SDAccel repository.

rachitnigam, 2020-08-12:
We published a paper and evaluated Dahlia!

closed 2020-08-12 by rachitnigam