cucapra/dahlia #27: Reading list

rachitnigam, 2018-09-18:
Tracking various relevant papers and articles.

rachitnigam, 2018-09-18:
FPGA Programming for the masses: https://dl.acm.org/citation.cfm?id=2436271

- "Writing an application at a behavioral level and leaving its destiny in the hands of a synthesis tool is commonly considered a bad design practice."

sampsyo, 2018-09-18:
Nice! Here are a couple more overviews in a similar vein:

* [Best-Effort FPGA Programming: A Few Steps Can Go a Long Way](https://arxiv.org/abs/1807.01340), from Jason Cong et al. This paper has a handful of case studies showing how to manually apply a small set of hardware-focused optimizations to an ordinary (software) program using HLS directives to get pretty good performance.
* [Parallel Programming for FPGAs](https://arxiv.org/abs/1805.03648), which is a whole book about using Vivado HLS to get high-performance designs.

rachitnigam, 2018-09-18:
High level languages for FPGA programming:
- Chisel
- Spatial
- [Kiwi](https://www.microsoft.com/en-us/research/wp-content/uploads/2008/04/kiwi_fccm2008.pdf)
- [Esterel](https://www.sciencedirect.com/science/article/pii/S157106610480820X)
- IBM Liquid Metal

rachitnigam, 2018-09-19:
Pragmas: Brief descriptions and issues that our type system can solve.
- `dependence`: "IMPORTANT!: Specifying a false dependency, when in fact the dependency is not false, can result in incorrect hardware. Be sure dependencies are correct (true or false) before specifying them."

rachitnigam, 2018-09-28:
Formal semantics for HDLs:

- [The semantic challenge of Verilog HDL](https://www.cl.cam.ac.uk/archive/mjcg/Verilog/V.pdf)
- [SecVerilog](www.cs.cornell.edu/projects/secverilog/)
- [Kami](plv.csail.mit.edu/kami/)
- [A Formal Executable Semantics of Verilog](https://www.ideals.illinois.edu/bitstream/handle/2142/17079/memocode-tech-report.pdf?sequence%3D2)

rachitnigam, 2018-10-01:
Types! Capabilities! Greek symbols!

- [Calculus of Capabilities](https://www.cs.cornell.edu/talc/papers/capabilities.pdf): Surprisingly easy read, with good explanations for the typing rules. Two major things to take away from this paper: (1) We need to implement regions and region inference for the "scope system" (where arrays can be consumed multiple times). The paper claims that this algorithm already exists, but I haven't looked into it. (2) If we want to have polymorphic in the arrays, the capability system will also need to be polymorphic. This makes the type system quite a bit more complex since the capabilities have complicated subtyping relationships that are easy to write in math but hard (?) to implement.

rachitnigam, 2018-10-08:
http://halide-lang.org/: Halide is a programming language designed to make it easier to write high-performance image and array processing code on modern machines.

Halide has some interesting abstractions for fast array and image processing. Worth taking a look at it to see if we can learn some interesting abstractions for seashell.

sampsyo, 2018-10-09:
On Halide:

- The [original PLDI paper](http://people.csail.mit.edu/jrk/halide-pldi13.pdf) is a joy to read.
- [Darkroom](http://graphics.stanford.edu/papers/darkroom14/darkroom14.pdf) is a Halide-like language for hardware generation from an overlapping set of authors.
- [Rigel](http://graphics.stanford.edu/papers/rigel/rigel.pdf) is a sort of sequel to Darkroom.
- There's also an unpublished paper on ["Halide-HLS"](https://arxiv.org/abs/1610.09405), which generates Vivado HLS input code from a Halide program.

sampsyo, 2018-10-09:
[Bonsai](https://dl.acm.org/citation.cfm?id=3158150) is the closest related work I know of at the intersection of type systems and program synthesis. Its goal is to find bugs in the soundness of type systems by synthesizing programs that are well-typed but fail at run time.

sa2257, 2018-10-12:
Vaguely relevant papers in OOPSLA and MICRO 18?

1. AnyDSL: A Partial Evaluation Framework for Programming High-Performance Libraries
https://2018.splashcon.org/event/splash-2018-oopsla-anydsl-a-partial-evaluation-framework-for-programming-high-performance-libraries
https://anydsl.github.io

2. Rethinking the Memory Hierarchy for Modern Languages
https://www.youtube.com/watch?v=XDpttL5_JIQ
http://people.csail.mit.edu/sanchez/papers/2018.hotpads.micro.pdf

3. TAPAS: Generating Parallel Accelerators from Parallel Programs
https://www.youtube.com/watch?v=Z6p-hClfg8k

And our own
4. An Architectural Framework for Accelerating Dynamic Parallel Algorithms on Reconfigurable Hardware
https://www.youtube.com/watch?v=SKMGP3VaPWY
https://www.csl.cornell.edu/~tchen/files/parallelxl-micro18.pdf

rachitnigam, 2018-11-26:
[Best Effort FPGA programming](https://arxiv.org/pdf/1807.01340.pdf)
- A study to demonstrate that HLS-based FPGA can be easy? Haven't convinced myself of this yet, but might be an interesting user-study approach.

[Generating Configurable Hardware from Parallel Patterns](http://www.cs.cornell.edu/~cdesa/papers/asplos16_prabhakar.pdf)
- Compiling functional parallel patterns to hardware. Has several relevant citations to other imperative PLs that compile to hardware in section 2.

rachitnigam, 2018-11-26:
https://hipacc-lang.org/.

Another Halide-like programming language for writing image processing kernels. The [halide-HLS](https://arxiv.org/pdf/1610.09405.pdf) paper points out that HIPAcc generates C which it feeds to Vivado HLS to generate FPGA designs. Also, Halide-HLS itself seems to use this workflow.

rachitnigam, 2019-02-07:
NESL programming language: http://www.cs.cmu.edu/~scandal/nesl.html

sa2257, 2019-02-07:
Papers Nitish shared from his work with Intel HLS on T2S, a language like Halide for spatial architectures. 
https://arxiv.org/ftp/arxiv/papers/1711/1711.07606.pdf

rachitnigam, 2019-02-13:
From Xilinx manual:
"A common issue when designs are first synthesized is report files showing the latency and interval as a question mark “?” rather than as numerical values. If the design has loops with variable loop bounds Vivado HLS cannot determine the latency"

We should be careful when implementing #63.

rachitnigam, 2019-02-14:
Matlab HDL: https://www.mathworks.com/products/hdl-coder/features.html#generating-hdl-code

rachitnigam, 2019-02-26:
[EASY: Efficient Arbiter SYnthesis from Multi-threaded Code](http://cas.ee.ic.ac.uk/people/gac1/pubs/JianyiFPGA19.pdf)

The authors created a new pass for the LegUp-LLVM tool chain that allows them to eliminate unnecessary [arbiters](https://en.wikipedia.org/wiki/Arbiter_(electronics)) from a hardware design. LegUp has a simple, conservative method to figure out how many arbiters a specific design and memory partition scheme needs. The paper uses the MSFT Boogie verifier to prove unique access to memories by different threads (which seem to be equivalent to unrolled loops). When the tool can prove unique access, it eliminates the corresponding arbiter connection from the design.

The technique is not precise, i.e., it cannot find the optimal arbitration scheme and relies on SMT to prove uniqueness which might take arbitrarily long. 

**UNVERIFIED**: Our type system is probably more precise than what they have since we support a strict subset of iteration patterns.

rachitnigam, 2019-03-14:
[SOAP: Structural Optimization of ArithmeticExpressions for High-Level Synthesis](http://cas.ee.ic.ac.uk/people/gac1/pubs/XitongFPT13.pdf): Haven't read yet. They claim:

> For  the  first  time,  we  bring  rigorous  approaches  fromsoftware static analysis, specifically formal semantics and abstractinterpretation,  to  bear  on  source-to-source  transformation  for high-level  synthesis.

rachitnigam, 2019-03-25:
[HardCaml](http://www.ujamjar.com/hardcaml/): Jane Street's library for producing FPGA designs in OCaml.

rachitnigam, 2019-03-27:
[FPGAs for the Masses](https://infoscience.epfl.ch/record/218533/files/EPFL_TH7004.pdf). Section 2 has a nice introduction the state of HLS tools. Note that this is different from a similarly titled paper above.

rachitnigam, 2019-06-17:
[Silica](https://github.com/leonardt/silica): Language from Stanford Pat Hanrahan's group. The basic idea is taking python co-routines and turning them into FSMs that execute up to each `yield` in the program control flow. Essentially, a `yield` acts like a cycle boundary and the compiler can schedule all the other co-routines to execute cooperatively every cycle.

rachitnigam, 2019-07-03:
[Reconfigure.io](http://docs.reconfigure.io/) is building a Go based HLS-as-a-service tool. Their compiler, Rio, uses LLVM to do the usual HLS scheduling stuff. The subset of Go used focuses on using channels and go routines. Tool seems beta.

rachitnigam, 2019-08-02:
Thesis on optimizations for HLS optimizations. Papers worth reading to understand semantics for loops in HLS. Haven't read them myself.

https://github.com/Junyi-Liu/benchmarks-HLS

sa2257, 2019-08-16:
Some papers which try to avoid complexities of HLS
- https://doi.org/10.1109/FCCM.2018.00029
uses machine learning to predict actual results from HLS reports
- http://delivery.acm.org/10.1145/3270000/3268344/a23-liu.pdf?ip=128.84.127.204&id=3268344&acc=ACTIVE%20SERVICE&key=7777116298C9657D%2EB493315FA1EC298D%2E4D4702B0C3E38B35%2E4D4702B0C3E38B35&__acm__=1565968561_27f17fae045b42e44c4dc67dbc0f5380
proposes a new framework to better map HLS to FPGA
- Best effort programming paper uses https://ieeexplore.ieee.org/stamp/stamp.jsp?tp=&arnumber=7446058 to claim HLS is hard. But it's not very convincing IMO.
- http://www.csl.cornell.edu/~zhiruz/pdfs/lamda-fccm2019.pdf talks about how hard DSE is

sa2257, 2019-08-16:
Adding in Bluespec reference manual http://www.bluespec.com/forum/download.php?id=157

rachitnigam, 2020-08-12:
This is now embedded in https://rachitnigam.com/files/pubs/dahlia.pdf

closed 2020-08-12 by rachitnigam