cucapra/dahlia #32: Buildbot tracker

sampsyo, 2018-10-03:
Here are some things we need to do to the buildbot:

* [ ] Finish fixing the Dockerfile to work with opam2 so it can actually run on Gorgonzola again. (@sampsyo is working on this.)
* [ ] Merge the workproc branch described in https://github.com/cucapra/seashell-gorgonzola/issues/1 to fix reliability issues.
* [ ] Make the buildbot actually run things on the Zynq board. (@sa2257 is working on this.)
* [ ] Add a no-Seashell mode to just HLSify and execute plain C programs.
* [ ] Add an option to just run the Vivado hardware resource estimation instead of doing the full synthesis flow.

sampsyo, 2018-10-03:
And here's a pie-in-the-sky idea:

* [ ] Get the buildbot running on an [Amazon F1](https://aws.amazon.com/ec2/instance-types/f1/) instance, which includes [Xilinx SDSoC for free](https://github.com/aws/aws-fpga/blob/master/SDAccel/README.md). This would get us access to a somewhat bigger FPGA and make the setup accessible to other people.

sa2257, 2018-11-07:
- [ ] Fix Areesh file copy bug
- [ ] Submit jobs with multiple hardware functions
- [ ] Consistently create an HLS error that seems to terminate the synthesis in some jobs
- [ ] Decide where to include library files/ why it throws a bug only in seac generated files
- [ ] Run parallel hls jobs
- [ ] Figure out licensing 
- [ ] Extract data out and see a synopsis in buildbot interface
- [ ] Prioritize jobs

sa2257, 2019-02-18:
- [ ] Option to change clock frequency. Currently running for fastest clock, but `gcd` is failing timing. Need to check whether a lower clock will help.
- [ ] Access to `Areesh` is failing. Check!!

sampsyo, 2019-03-10:
Some quick context on the new additions from @rachitnigam:

> Generate job name based on the `cpp` or `fuse` file in the provided archive.

This is trickier than it sounds because the job name is assigned before we extract the archive.

> Also the time currently given is 3 hours late.

Time zones are hard. :)

> Change job view to hide most build files under a fold and only focus on `rpt` files and archive files

This is the same idea as the second item in the previous chunk. These should be on separate pages, like `jobs/<ID>/log.txt` and `jobs/<ID>/files/`, linked from the main job description page.

rachitnigam, 2019-03-18:
Added: Buildbot should detect if the archive contains a `Makefile` and run `make` to generate `<kernel>` file. This is help in the case of machsuite since it needs to compile several `.cpp` files to `.o` and link them to generate the host function. The call to `make` can set an environment variable that Makefiles can use to generate the `.o` files.

sa2257, 2019-03-19:
Regarding the `Makefile`. I don't see why that shouldn't be used to compile software sources for host. But using `make` for hardware functions might make Buildbot oblivious to hardware functions in a job. For one, this might make extraction harder. 

The underlying issues seem to be,
1. Buildbot assumes functions other than main are hw
2. Buildbot assumes hardware function name to be same as file name (so that `hw_basename` can be used throughout)
3. Uncertain what happens when multiple functions are in hw.cpp, this works in theory (SDSoC handles it), but Buildbot might not support it. (Running a test to confirm)
4. At least the hardware function compilation cannot access common header files or local header files through `.cpp` or `.h`. Main function can access these, I assume software compilations should also be fine.

The workaround was,

1. - name host as main and that was the only sw
2. - name hw function the same as file name
3. - no header files used in hw function compile

But these (3) seem to be too restrictive to run some MachSuite apps (fft/sort) and (1/2) are too restrictive to run MachSuite as is.

rachitnigam, 2019-03-19:
Yeah, I understand the issue you mentioned. It seems that sds does in fact handle *simple* header includes for the hardware function. For example stencil_3d includes a header file to define types.

A different way of doing this would be having some sort of configuration sent to buildbot that specifies all the software dependencies for the host and have buildbot manually run those commands for us.

The way machsuite seems to be setup seems to make this model of separate compilation and linking necessary so (1) and (3) need to be fixed. (2) doesn’t seem to be a massive issue.

sa2257, 2019-03-19:
Buildbot can run multiple hardware functions with separate cpp for each function [job](http://gorgonzola.cs.cornell.edu:8000/jobs/aRvlGACQ0QY.html)

Buildbot can run with all functions in the same file [job](http://gorgonzola.cs.cornell.edu:8000/jobs/Kp6wLP3Q7N0.html)

Buildbot seems to synthesize (check sds_mmultadd.rpt) function calls within hardware function, but doesn't generate an estimate [job](http://gorgonzola.cs.cornell.edu:8000/jobs/rhe0k9GzPxQ.html)

sa2257, 2019-03-19:
In light of this, it might just be enough to use a `Makefile` to provide software dependencies. We can go with 2 for now. 1 and 3 have to be addressed, 1 shouldn't break the current flow.

rachitnigam, 2019-03-19:
I'm not sure I understand when (3) breaks. The stencil_3d example compiles fine with a header file included in the C++ file.

sa2257, 2019-03-19:
> I'm not sure I understand when (3) breaks. The stencil_3d example compiles fine with a header file included in the C++ file.

Can you post a relevant job? (Unless it's too hard, I can run it again)

rachitnigam, 2019-03-19:
Added @sa2257's issue: #146

rachitnigam, 2019-03-19:
Added "Job page should extract all log lines that have the word "warning". These usually seem to indicate when a particular pragma is being ignored."

rachitnigam, 2019-03-19:
Added "Run task should upload required data files to Areesh."

rachitnigam, 2019-03-28:
Added: First order support for Amazon F1 instances.

It occurred to me that we don't need to completely port the buildbot infrastructure to Amazon instances in order to use larger boards. We can do this incrementally:
1. We can run synthesis on gorgonzola and set the hardware target to the board on the Amazon instance. Then, we can change the `fpga_executing` stage to upload it on the instance and run it there.
2. Next, we can change the `onxilinx` script to instead send our data to the instance and run synthesis on F1 and wait for results. Note that this composes nicely with 1.

sampsyo, 2019-03-28:
True! FWIW, I bet that running the synthesis step (i.e., running `sds++`) on an F1 instance will actually be pretty straightforward—if the whole thing is running there, it should be just as "easy" as changing the hardware target. (No `onxilinx.sh` script required.)

It's probably not worth a checkmark here because it's longer-term, but if/when we all get access to Intel's vLab infrastructure, I'd be really interested to see how hard it would be to generate code for their HLS tool too.

rachitnigam, 2019-03-28:
Oh, I've already gotten access from Intel.

sampsyo, 2019-03-28:
Awesome. The first stumbling block I noticed is that [their documentation](https://wiki.intel-research.net/FPGA.html) is all about OpenCL-based synthesis, not "C with pragmas" as input. So that could be kind of annoying to generate.

PR #248 (2019-08-24): Move buildbot to separate repo
Fixes #210.
Fixes #32.
closed 2019-08-24 by rachitnigam