cucapra/dahlia #37: Sized ints overflow

tedbauer, 2018-10-23:
For the current implementation of sized integers, huge numbers are stored in the index types, and with bit numbers as high as 32 or so there are overflow errors. It seems a bit silly to store such big numbers, so maybe we could change the `TIndex` representation for the implementation? Would need to think about how to differentiate static ints I think.

sampsyo, 2018-10-23:
Good point. One trick might be to store, for the range endpoints, _either_ a plain number _or_ a power of two:

```
type Offset = Lin of int | Exp of int
```

so `Exp 32` represents the number 2^32 while `Lin 32` represents plain old 32.

rachitnigam, 2018-12-03:
@tedbauer is this fixed?

rachitnigam, 2019-01-28:
@tedbauer ping

rachitnigam, 2019-02-06:
Implemented in #46

closed 2019-02-06 by rachitnigam