cucapra/dahlia #41: Implement %{ }%

rachitnigam, 2018-12-03:
`%{ stmts }%` allows the program to recover the capabilities for arrays and imply that a sequence of statements are executed on a different clock cycle. Implementing this for now requires changing the capability checking and type checker.

This program is valid:

```
%{ a[i] := c[i] + 1 }%
%{ b[i] := 2 * a[i] }%
```

sa2257, 2019-03-02:
Hi, unsure in which issue I should request this clarification.

I presume this is the initial discussion on the serial operator `---`

Is the serial operator intended to sequence memory operations into different clock cycles?

It's not the same notion of sequencing between the `combine` operator scopes I presume?

```
let x = 1;
let y = x;
```

This doesn't need sequencing with `---`, does it?

rachitnigam, 2019-03-02:
Yes.

commit 66f18680 (2019-01-31): Add refresh banks. Fix #41
commit e5c12955 (2019-02-06): Add refresh banks. Fix #41