cucapra/dahlia #47: Meaning of refresh banks in combine blocks

rachitnigam, 2019-02-04:
What does this piece of code mean?
```
decl a: bit<16>[10 bank 2];
for (let i = 0..10) unroll 2 {
  let x = a[i];
} combine {
  sum += x;
  ---
  a[i]
}
```

Note: `---` is the new syntax for `%{ }%`

rachitnigam, 2019-02-04:
Well, that's a different problem that I wasn't focusing on. In the implementation, a call to `---` returns all banks to all arrays. I'm asking if this should be the case when `---` is used inside a combine block.

sampsyo, 2019-02-04:
Aha! Probably, so that this is allowed?

```
for (whatever) {
    irrelevant code;
} combine {
    b[0];
    ---
    b[0];
}
```

rachitnigam, 2019-02-04:
Yeah, but it'll also allow:
```
for (i ...) {
  let x = a[i]
} combine {
  ---
  a[0] += x
}
```
which *might* be incorrect?

commit 8f426f5d (2019-02-05): fix #47
PR #46 (2019-02-06): Scala rewrite
Fixes #44 #41 #37 #30. Subsumes #40 #43.

**TODO**
- [x] Update buildbot scripts
- [x] Write new Dockerfile
- [x] Explicit typed let
- [ ] ~views~
- [x] reductions
- [ ] capabilities
- [ ] iterator stack checking (#50)
- [x] Fix #47 
- [x] subtyping review
closed 2019-02-06 by rachitnigam