cucapra/dahlia #51: Type checking rule for `---`

rachitnigam, 2019-02-06:
We have the following type checking rule for `e1 --- e2`.

```
G, e1 --> G', t1    G, e2 --> G'', t2  Gf = G' + G''
---------------------------------------------------------------    where G' + G'' = G''
                G, e1 --- e2 --> Gf, t2
```

The definition of `+` on `G` is incorrect for this case:
```
{
  let x = 1;
  ---
  let y = 2;
};
x + y // error with `x` is unbound.
```

Figure out the right `+` and implement it.

sampsyo, 2019-02-06:
As a starting point, it might be worth looking around at other papers that use a linear or affine type system in an _imperative_ language. The usual formulation for the lambda-calculus does not have "result contexts," which are necessary to support imperative-style variable definitions. Those papers doubtless need to figure out, for `;`-style sequential composition, how to both support "building up" a context with declarations while "dividing up" contexts when checking both sides of this composition.

(I don't have any specific paper recommendations right now, but I may be able to dig some up…)

rachitnigam, 2019-03-03:
This program doesn't work:

```
let bucket_idx = 0;
---
bucket_idx := 10
```

apurvakoti, 2019-03-07:
Just trying to get some clarity on the motivation behind the original typechecking rule. If `G' + G'' = G''` then evaluating `e1 --- e2` under `G` is essentially evaluating `e2` under `G`, as if `e1` was never there.
In addition, having `Gf` be the intersection of `G'` and `G''` (as in the comment in TypeCheck.scala) wouldn't help the `bucket_idx` example.

As far as I can tell, wouldn't the simplest solution be evaluating `e2` under `G'` and _then_ allowing the resulting environment to be `G''`? Want to make sure I'm not misunderstanding the purpose of `---`.

rachitnigam, 2019-03-07:
No, the original type checking rules is just plain wrong.

We want a few things: 

1) — should “recover” the banks between e1 and e2. Concretely, this can be implemented as checking both of them in the original G to get G1 and G2.

2) anything in parallel sequence with e1 — e2 should not be able to use anything used by the merge of G1 and G2.

3) any declarations from G1 should be available in G2. 

Something that’ll help is writing down a few examples and asking if they should or shouldn’t typecheck.

For an implementation, we’ll probably have to use endScope after e1 is checked to get any new bindings B and add them to the G. This means e2 is checked under G + B.

sampsyo, 2019-03-07:
I was thinking a little bit about what makes this hard today and doodled a few notes:

[image]

I was trying to sketch the rules for any generic composition operator (think of `;` as being some composition operator for any language, not necessarily our `;` or `---`).

In the upper-left, I wrote my impression of how these rules usually go in "normal" functional languages. In the lower-left, I showed how this has to change in a substructural type system. The "U with plus" operator is a disjoint union on contexts, so Gamma1 and Gamma2 have to be two "pieces" of the overall Gamma—and together cover all of Gamma.

In the upper-right, I sketched how the rule usually looks for checking composition of statements in an imperative language. I probably should have used "c" instead of "e" to remind us that we're dealing with commands. The idea is that the form of the judgment needs to change so you can build up a "result context," which might include newly-declared variables. So we have something like:

```
\empty |- let x = 5 -| { x }
```

That is, the rule for declarations needs to "add" something to the context. The composition rule I've shown in the upper-right uses Gamma' to check e2 because it might need to use declarations from e1.

The lower-right is meant to show why designing a substructural version of the imperative composition rule is hard. The question is, what do you use as the input context for checking e2? On the one hand, you want to use Gamma2 because that's how you "divide up" Gamma. On the other hand, you want to use Gamma', because that's how you pick up new declarations that appear in e1. Neither one really suffices!

It seems like what the rule needs is some of both: get "new" declarations from Gamma' and "old" declarations from Gamma2. But figuring out what's "new" and "old" in each context seems pretty messy…

rachitnigam, 2019-03-08:
@apurvakoti @sampsyo and I talked about this and we came to the decision that the source level typechecking should use `addScope` and `endScope` to extract the new bindings added in `e1` and use add them to the type checking for `e2`.

The type checking rule for the core calculus will not thread through variables in this magical way. If a `let` binding needs to be used in both the branches, the code should write this:

```
let x = 1;
{
  x := 10;
  ---
  x := 11
}
```

The source level type checking rule can be viewed as doing this "hoisting" implicitly.

apurvakoti, 2019-03-08:
Ok, I can work on an implementation for this.

PR #122 (2019-03-14): Fixing CSeq type checking rule
Fixes #51 
Creates a new scope in the environment; typechecks `c1`; extracts any new bindings from this typechecking; adds the new bindings to the initial environment and typechecks `c2`.
closed 2019-03-14 by rachitnigam