cucapra/dahlia #54: Functions implementation sketch

rachitnigam, 2019-02-10:
Proposal to bring back functions:

- Create new `CFuncDef` and `EApp` nodes to represent function definitions and applications. Change `Prog` to be a list of declarations, followed by a list of function definitions, followed by top level commands.
  - Alternatively, we can try removing `decl`s altogether and require a `main` function to be defined.
- For typechecking, extend the environment to map function identifiers to `CFuncDef` nodes. Checking `EApp`s is non-trivial since memories used in an `EApp` expression need have their banks consumed correctly. Follow this algorithm
  - Check of if the actual parameters satisfy constraints on formal parameters.
  - Typecheck the function body using formal parameters and "write back" the bank consumptions to actual paramters.
  - **Optimization**: Pre calculate these consumptions instead of re-running for each EApp.
- Extend `Emit` to spit out function definitions.

sampsyo, 2019-02-10:
Sounds good!

> Alternatively, we can try removing `decl`s altogether and require a `main` function to be defined.

Even more simply, the structure could be like C: the program could just be a list of functions—without a requirement that one of them be named `main`. If you want to run the program independently, then it surely needs a `main`, but having `main` only makes sense if we're not actually using the FPGA. For actual FPGA execution, `main` will always need to be defined in a separate module (written in plain C).

> Typecheck the function body using formal parameters and "write back" the bank consumptions to actual paramters.

This amounts to an interprocedural analysis. The optimization you mention—where we calculate "signatures" for functions once to avoid re-analyzing function bodies on every call—could turn this into an intraprocedural check, which would be nice but sounds pretty hard.

A simpler intraprocedural step would be for function calls to consume _all_ of their arguments, i.e., to conservatively assume the function will need _all_ of the passed-in arrays' banks. Although simple, I suspect this will suffice for most cases—and it will make it possible for programmers to reason about functions independently.

rachitnigam, 2019-02-10:
- Re: Intraprocedural analysis, I like the idea of a conservative analysis. It probably suffices for a first cut.
- The point about a `main` function is in anticipation of polymorphism. The sketch for polymorphism in my mind is building a DAG (since recursion and use before define are not allowed) of all functions and then inferring the right parameters for each function. The utility of a `main` function is that it serves as a function that it only uses concrete values.
  - OTOH, there is no reason the last function defined can't have this restriction.

sampsyo, 2019-02-10:
Got it—so the type inference algorithm would like to have an "entry point." That makes sense! Especially because we can't _emit_ polymorphic code (i.e., we need to have concrete parameters for everything before codegen).

PR #67 (2019-02-13): Function support
Fixes #54 #62.

- Program structure is now a list of function defs followed by a list of decls, followed by top level commands.
- Applications consume array expressions completely
- Functions cannot return values (will potentially fix this in the future) and have return type `void`.
closed 2019-02-13 by rachitnigam