cucapra/dahlia #56: Loop carry dependencies from local variables

rachitnigam, 2019-02-10:
This program should be rejected since it is creating a loop carry dependency. The sequential and parallel semantics don't match up:
```
let c = 0;
for (...) unroll 3 {
  c := a[i] + c
}
```

However, this example is fine:
```
for (...) unroll {
  let c = 0;
  c := a[i] + c;
}
```
since `c` cannot carry dependencies over.

Adrian says:
> I think the right restriction is probably something like “variables declared outside a loop can be read or written in a loop body, but not both”

To figure out a solution, should these FAIL/PASS:
(1)
```
let c = 0;
for (...) unroll 3 {
  c := 1;
  a[i] := c;
}
```

(2)
```
let c = 0;
for (...) unroll 3 {
  a[i] := c
}
```

(3) **Note the missing `unroll`**
```
let c = 0;
for (...) {
  c := a[i] + c;
}
```

(4) add unroll to (3)
```
let c = 0
for (...) unroll 3 {
  c := a[i] + c;
}
```

rachitnigam, 2019-02-20:
Implement this as a separate compiler pass to do a "Definition use analysis". The semantics are as follows:

- In the parent scope, a variable can be both read and written to.
- In a scope that inherits a binding, only one of read or write is allowed.
  - For the purposes of this check, `for` body and `combine` body count as a single scope.

sampsyo, 2019-02-21:
Sounds good! For what it's worth, it looks like the Wikipedia article about def/use chains is pretty good:
https://en.wikipedia.org/wiki/Use-define_chain

rachitnigam, 2019-02-22:
Braid has some code for this: https://github.com/cucapra/braid/blob/master/src/compile/defuse.ts

tissue3, 2019-10-02:
The implementation idea is following:
All variables are tracked with a <name, state> map. At first, the whole program is scanned to learn the names, but all states are empty.
Whenever a define or use is encountered, the state will change. Whenever "---" is encountered, the state will get cleared for all names.
If else statement will be treated as a whole. That is, in the case of 
if(y){
  x = x+1;
}else{
  x=1;
}
we will simply ban it.

tissue3, 2019-10-02:
The indexing statement is not clear to me though. 
while (i > 0) {
  a[i] := i;
  i := i - 1;
}
Shall I ignore all indexing? What is the expected hardware behaviour of a[i]: = i?

tissue3, 2019-10-02:
Another question is for nested loop. Shall I assume nested loops are sequential? Shall the code below be allowed
for (let i = 0..10) {
  y := 1;
  for (let j = 0..10){
    a[j] := y;
  }
}

tissue3, 2019-10-03:
Also, I am lost in scala. Shall I define an object like ScopeMap to store the <name, state> pair?

rachitnigam, 2019-10-03:
> The indexing statement is not clear to me though.

`while` loops are not unrollable so that case doesn't matter that much. In `for` loops, if you try to do:
```
for (let i = 0 .. 10) {
  i := i + 1;
  a[i];
}
```
it should report an error.

> Another question is for nested loop. Shall I assume nested loops are sequential? Shall the code below be allowed?

If the inner loop in unrolled, yes. If the outer loop is unrolled, no.

tissue3, 2019-10-06:
Would this be illegal when checking loop dependencies since the combine statement is by default unrolled?

```
for (let i = 0..6) unroll 3 {
  for (let j = 0..8) unroll 4 {
    let rs = M[i][j] * V[j];
  } combine {
    R[i] += rs;
  }
}
```

tissue3, 2019-10-07:
My ideas about implementation:
1. I want to embed loop checking into type checking because the general frameworks are the same. There would be a lot of repeating codes if they are separate.
2. Define -> use inside a loop is fine. But use->define is not, so the rule is execution time. First use in a loop will depend on last define. I might need two maps firstUseMap<Id, execution line 1> and lastDefMap<Id, execution line 2>. At the end of the loop scope, check if #line 1 < #line2. If not, report error.
3. If statement is so confusing. In the following case, which one is defined, `x` or `y`?
```
if(statement){
  x := 1;
}else{
  y := 1;
}
```

sampsyo, 2019-10-08:
1. Cool—I don't have a strong feeling about this, but you might find it's actually easier to write your own visitor instead of just injecting things into the type checker. Than you won't need to deal with passing type environments around, like the type checker must. And I think you'll find that @rachitnigam's `Checker` class makes it pretty easy to do all the boring tree traversal stuff without writing much code of your own.
2. Getting line numbers to do this actually seems really hard! Remember that a definition and a use might be on the same source line. The ordering we actually care about is AST ordering, not source-code line ordering. So instead of trying to maintain maps to keep track of the positions, how about this: just keep track of a list of the currently defined variables. Then check every use, and make sure that it's defined already (i.e., in that list). You could use a second pass to allow variables that are _only_ read inside the loop and _never_ written.
3. Both are defined! That is, this is a conservative analysis, so we'll need to assume that `x` and `y` are both written to.

tissue3, 2019-10-09:
New implementation:
1. There are fours kinds of def-use relation of a varaible in a loop:
| Chain     | error or not |
| --------- | ------------ |
| Use->Def  | should error |
| Def ->Def | fine         |
| Use ->Use | fine         |
| Use ->Def | fine         |
So we would like to keep a map logging Def/Use states.
For example, `a=0; c = b+1` adds three terms in the table
| var  | def/use |
| ---- | ------- |
| a    | def     |
| b    | use     |
| c    | def    |
At each def of the variable, say `b`, we check if `b` is defined or used before. If it's used, report error; if it's defined, do nothing; if it's not in the table, insert `b->def` to the table.

2. Now consider the `if` statement. `if` introduces the third state, don't-know state, dk for short. 
```If cond
  y=0
x = y
```
should report an error because `x` is defined by `y`, but we don't know if `y` is defined in the same loop. That is, `if` statement damage the `Def -> Use` chain. We need to add a new state as a complementary to define. Let's put new relation into the table. Notice `x`->DK can only be inserted when `x` is the lhs of an equation.
| --------- | ------------ |
| Use->DK   | should error (use of variable inside `if` is also forbiddened) |
| DK->DK    | fine                                                         |
| DK->Use   | Error                                                        |
DK and Def are two mutable states. That is, they can updated each other. Consider
```
If cond
 y = 0
y = 1
x = y
```
In this example, `y` changes state from DK to Def and `x=y` is fine. So finally, we can add two more colomn to the replation
| --------- | ------------ |
| Def->DK   | fine (update Def to DK) |
| DK->Def    | fine (update DK to Def) |

3. Nested for loop with unrolling. Because increasing of unrolling factor is monotonic, it should be fine to keep just one map for all relation ship. That is, updating the outer loop always updates the inner loop. However, for the easiness to implement `if scope1 else scope2`, I will use scopeMap, so we can merge two scopes with DK states.

sampsyo, 2019-10-09:
Sounds great! Here are some incoherent notes that I will explain more in person:

---

- In the first table, I think the last row should say "Def -> Use"?
- I think it would be sound to consider everything *not* in the table to initially be in the "def" state, right?

---

> Notice (x,DK) map can only be inserted when x is the lhs of an equation.

Just for fun: a compilers-y way of saying this is that "x is used as an lvalue in an assignment." :)

---

I don't entirely understand what's wrong with the "DK -> use" transition yet.

tissue3, 2019-10-09:
Oh, everything not in the table doesn't have any state. Let's say we enter a loop, `b=a+1`, `a` is used, it is not defined/used/dked before. It's fine. We inserted a term `(a,use)` in the table, so that next time we see it is defined inside the loop, we know we should error.
A example looks like this
```
for i ...
  b = a+1
  a = c
```
Thanks for the instructing my poor writing! Please always point out so I may get better.

> Sounds great! Here are some incoherent notes that I will explain more in person:
>
> * In the first table, I think the last row should say "Def -> Use"?
> * I think it would be sound to consider [...]

rachitnigam, 2020-02-08:
Fixed by #264, #279, and #280.

closed 2020-02-08 by rachitnigam