cucapra/dahlia #61: "And" type

tedbauer, 2019-02-13:
For some of the MacSuite apps, it would be helpful to have some kind of "and" type, like a `struct`. Would be helpful in particular in:
- `bfs/queue`: need a way to represent a node of a graph.
- `md/grid`: in the C code, has vectors represented as `struct`s.

rachitnigam, 2019-02-13:
From Xilinx manual: "Structs by default, are decomposed into their member elements. Each member element is implemented as a separate element. An array of structs, where each struct has M elements, is decomposed into M arrays. The DATA_PACK directive is used to flatten structs into a single wide-vector."

We can provide support for `structs` in two ways: (1) simply compile them to C `struct`, or (2) do something interesting with the above mentioned decomposition.

----

`DATA_PACK`: "Packs the data fields of a struct into a single scalar with a wider word width."

"The DATA_PACK optimization directive is used for packing all the elements of a struct into a single wide vector. This allows all members of the struct to be read and written to simultaneously. The member elements of the struct are placed into the vector in the order the appear in the C code: the first element of the struct is aligned on the LSB of the vector and the final element of the struct is aligned with the MSB of the vector. Any arrays in the struct are partitioned into individual array elements and placed in the vector from lowest to highest, in order."

"**The DATA_PACK, ARRAY_PARTITION and ARRAY_RESHAPE directives are mutually exclusive**"

sampsyo, 2019-02-13:
Interesting! For what it’s worth, this transformed representation:

> An array of structs, where each struct has M elements, is decomposed into M arrays.

has a name: “struct of arrays” (SOA), as opposed to the typical (C) representation, which would be “array of structs” (AOS). An AOS-to-SOA transformation is usually (but not always!) a good idea for locality during iteration.

rachitnigam, 2019-02-18:
After discussion with @sampsyo, we've decided that the first pass implementation will simply compile record style types into plain `structs`.

However, we should note that an AOS (array-of-struct) representation is more flexible when it comes to indexing. We might support a more flexible type checking rule for AOS struct representations.

rachitnigam, 2019-02-21:
Records allow for aliasing to the same array. Consider:
```
record points {
  a: bit<32>[2];
}
let k: points = { a = [1;2] };
let x = k;
let y = k;
x.a[1];
y.a[1]; // No error
```
The right thing to do is probably giving `k` a linear type and consuming it on a read. @sampsyo thoughts?

sampsyo, 2019-02-21:
The C semantics for this actually would not have any aliasing—the array field is part of the struct value. So the assignments to x and y create copies of the entire value, including the array it contains. To get two structs to alias the same underlying array, you'd need to use pointers.

I'd suggest that our language should probably behave about the same. That is, if structs are going to be allowed to contain arrays (which is not a foregone conclusion!), then those arrays should not be memories. If they were, then you could only ever have a fixed number of structs of a given type. That is, you could not have an array containing those structs, because you'd need an unbounded number of memories to implement the arrays they contain.

rachitnigam, 2019-02-21:
> That is, if structs are going to be allowed to contain arrays (which is not a foregone conclusion!), then those arrays should not be memories
What's the alternative?

sampsyo, 2019-02-21:
They would (again, if allowed) would just be fixed-size array values. An array of size N would work identically to N fields on the struct of the same type. As with any other type, each struct only resides in a single memory, so the array is just laid out in sequence in the memory region for the struct.

rachitnigam, 2019-02-21:
I see. Should I punt on allowing arrays in structs for now? @tedbauer would that work for your example? I see that the parameter to the input function in md is a struct, but that simply contains a bunch of arrays.

sampsyo, 2019-02-21:
I’d say yes, if we can get away with it—aliasing is a can of worms we likely don’t want to open!

tedbauer, 2019-02-21:
Yeah, I think that would work! The programs I had in mind were `bfs` and `md`, which just have structs that have a few integer fields. (Graph nodes in `bfs` and 3D vector with x, y, z in `md`.)

PR #83 (2019-02-23): Struct support
Fixes #61.

- [x] Restrict usage of arrays in structs
- [x] Allow writing struct literals.
closed 2019-02-23 by rachitnigam