cucapra/dahlia #63: While loop support

rachitnigam, 2019-02-13:
Can we support dumb while loops that don’t support any unrolling?

@sa2257 what does HLS do with while loops?

rachitnigam, 2019-02-13:
See https://github.com/cucapra/seashell/issues/27#issuecomment-463287285 .

rachitnigam, 2019-02-13:
`LOOP_TRIPCOUNT`: "Used for loops which have variables bounds. Provides an estimate for the loop iteration count. This has no impact on synthesis, only on reporting"

Might help with above. If our `while` loops are bounded, we might be able to do interesting things.

sa2257, 2019-02-13:
Good question. I’ll report after trying it out! I don’t have a good idea how they work or whether they have any contraints.

rachitnigam, 2019-02-18:
Ping @sa2257 any updates?

sa2257, 2019-02-18:
@rachitnigam HLS supports while loops.

The `tripcount` is just for estimations. It doesn't affect the synthesis. 

I don't really understand what you mean by "we should be careful when implementing variable bound loops" and "if `while` loops are bounded we can do something interesting"? But support `while` loops without unrolling makes sense. 

I'm running some examples. But running into backend issues.

rachitnigam, 2019-02-18:
> I don't really understand what you mean by "we should be careful when implementing variable bound loops"
It’s non trivial to figure out how many times a while loop runs. We should make sure that we don’t break the type system by introducing while loops.

> "if while loops are bounded we can do something interesting"?
If you can statically figure out the bound of a while loop with the iteration variable, you can compile it to a for loop.

> But running into backend issues.
Vivado or buildbot?

sa2257, 2019-02-18:
HLS just uses the pragma to give estimation values for hardware timing. I think that won't be very useful for us. Since we have types, we should be able to generate the `trip count` without user input pragma. This analysis might allow us to know what is the common path and optimize for that?

Running into timing issues in `vivado`. It'll be interesting to see what they generate.

The example I'm running is 

```
for (i = 0; i < N*N; i++) 
#pragma HLS unroll factor=ELE  
    Minim_Loop: while (A[i] != B[i]) {    
      if (A[i] > B[i])  
        A[i] -= B[i];  
      else 
        B[i] -= A[i];  
    }    
    C[i] = A[i]; 
}  
```

where you shouldn't be able to statically know the bounds?

I'm also getting an error if I partition the arrays in `i`. I don't see why though.

rachitnigam, 2019-02-18:
I understand what `TRIPCOUNT` does. Also, yes, that's what I was suggesting. If we have a while loop where we can infer bounds, we can emit the right tripcount pragma. 

It'll be helpful if you paste the error here. Might be worth talking about the example in the paper if it is something subtle.

sa2257, 2019-02-18:
Partition error? 
```
ERROR: [XFORM 203-103] Cannot partition array 'A' (/.../gcd.cpp:3): array access out of bound (/.../gcd.cpp:20:5).
ERROR: [HLS 200-70] Pre-synthesis failed.
```

Line 3 is the function definition.
Line 20 is `C[I] = A[i]`

I'm keeping track of the timing error elsewhere. Will add them if it's interesting to talk about.

rachitnigam, 2019-02-19:
Wait, this is really trivial, I think. You're unrolling the outer loop `N` times and creating `N` copies of the inner `while` loop. If `N` copies execute at the same time, you have a memory safety violation.

PR #75 (2019-02-19): While loops
 Fixes #63
commit e8dd388d (2019-02-19): support while loops; fixes #63.
closed 2019-02-19 by rachitnigam