cucapra/dahlia #64: Pipelining

sampsyo, 2019-02-13:
Without a specific proposal, this issue is to remind us that we will need some manner of pipelining annotation for loops. It's likely we can use a desugaring strategy similar to the new one for unrolling as described in #58, which takes a source loop like this:

```
for (...) unroll k {
  c1
  ---
  c2
}
```

To:

```
for (...) {
  c1(0); c1(1); ...
  ---
  c2(0); c2(2); ...
}
```

Where a pipelined version of the original loop would look something like this, where the "duplicated" commands get offset according to the initiation interval:

```
for (...) {
  c1(0)
  ---
  c2(0); c1(1)
  ---
  c2(1)
}
```

rachitnigam, 2019-02-14:
Page 208 of the manual says that when Vivado can't figure the value of a dynamic iterator, it assumes that all reads depend on write and vice-versa.

In this example:
```
for (row = 0; row < rows + 1; row++) {
  for (col = 0; col < cols + 1; col++) {
  #pragma HLS PIPELINE II=1
  if (col < cols) {
    buff_A[2][col] = buff_A[1][col]; // read from buff_A[1][col]
    buff_A[1][col] = buff_A[0][col]; // write to buff_A[1][col]
    buff_B[1][col] = buff_B[0][col];
    temp = buff_A[0][col];
  }
```
Vivado will not respect a `PIPELINE` pragma. The manual states that program has to manually specify the lack of dependency using `DEPENDENCE` pragma.

Since we are already very conservative with pipelining and don't allow bank conflicts, we should aggressively use this pragma.

rachitnigam, 2019-03-17:
While thinking about an implementation strategy, I came across this discussion:
https://forums.xilinx.com/t5/Vivado-High-Level-Synthesis-HLS/How-to-pipeline-a-for-loop-including-functions/td-p/456030

This comment in the thread is slightly concerning to me:
> When a loop or function is pipelined, Vivado HLS unrolls all loops in the hierarchy below the function or loop so as to meet your pipeline directive.

This would mean that some new syntax of the form:
```
for (let ...) unroll .. pipeline k {
  for (...) ... pipeline l {
  }
}
```

might not preserve the semantics we claim it has (where the inner loops has an initiation interval of `l` while the outer loops has `k`). I'm trying to find more in depth explanation of the pragma. The [explanation in Vivado manual](https://japan.xilinx.com/html_docs/xilinx2017_4/sdaccel_doc/fde1504034360078.html) isn't complete. @sa2257 do you know of any other sources I should read?

A higher level point is we need to clarify the semantics of pipelining in nested loops.

rachitnigam, 2019-03-17:
The [occurrence pragma](https://www.xilinx.com/html_docs/xilinx2017_4/sdaccel_doc/zah1504034366392.html) also seems important for pipelining things effectively. Not sure if there is a clean semantics for those.