cucapra/dahlia #70: Array access using dynamic ints

rachitnigam, 2019-02-14:
Some of the examples in the Xilinx manual allow for this:
```
void inp(int a[10], int readline) {
  int c = a[readline/2];
}
```
It seems we can conservatively allow for these by making dynamic accesses consume all banks of the array.

sampsyo, 2019-02-15:
It occurs to me that allowing dynamic access requires all-way muxing, i.e., if the array is banked, we need a view that is unbanked (unibanked). Which is the same hardware implementation and typing effect as for our `flex` view proposal (c.f. #57) in the special case where the result has only a single bank.

So the policy should likely be that unbanked arrays can be accessed with dynamic integers. (And you can use `flex` to get an unbanked array view if you need to.)

rachitnigam, 2019-02-18:
This sounds reasonable. Concretely, the `consumeBanks` function in `typeCheck` needs to change to allow for `TSizedInt` and fully consume the dimension. The case statement should also make sure that the dimension is unbanked and reject the access if it's not.

This can be done in parallel with the `flex` proposal.

closed 2019-02-28 by rachitnigam