cucapra/dahlia #73: Operations on index types not used for array access

rachitnigam, 2019-02-19:
Fuse disallows:
```
a[2 * i] // i is an iterator
```
Since `*` is not defined on index types. However, it will also reject this:
```
x[0] := 2*i; // i is an iterator
```

I think the latter should be allowed with the following semantics:
When an index type has an operation performed on it, it gets upcast to a sized int. This also has the nice property of making sure that the first program is still rejected (since access using dynamic ints is not allowed). 

The result of the operation is also a dynamic int.

@sampsyo if this sounds good, please LGTM.

sampsyo, 2019-02-19:
Yes!! I'm all for it.

This is less essential, but you could also consider an explicit conversion (`2 * int_of_idx(i)`, named in OCaml style, for example). The implicit conversion would just do the same thing "on demand."

rachitnigam, 2019-02-19:
Since the upcast function is all in one place, we can simply generate explicit annotations like this for every upcast is a restricted version of the language.

I'm imagining a `fuse --strict` flag that enforces all of these conversions explicitly.

PR #74 (2019-02-19): Allow arithmetic on idx types
fixes #73
commit 130f89a9 (2019-02-19): allow arithmetic on idx types; fixes #73
closed 2019-02-19 by rachitnigam