cucapra/dahlia #76: Else statement

tedbauer, 2019-02-19:
I keep finding myself writing this pattern:
```
if (cond) {
...
} 

if (!cond) {
...
}
```
Doesn't seem critical but would make my soul feel better if there was an `else` construct.

rachitnigam, 2019-02-19:
Ha, turns out `else` is also non-trivial. To type check `If(cond, c1, c2)`, we need do:
```
env1 = checkC(c1)(env)
env2 = checkC(c2)(env)
max(env1, env2)
```

The `max(env1, env2)` operator finds the maximum consumed banks between both branches and returns a new environment `e'` where for all memories, `e'(m).consumedBanks >= env1(m).consumedBanks` and `e'(m).consumedBanks >= env2.consumedBanks`.

Concretely, consider this example:
```
a: int[2 bank 2]
if (...) {
  a[0]
} else {
  a[1]
}
a[0] // unsafe. we don't know which branch executed
```

@sampsyo is there a well known conservative estimations of this in linear type systems?

rachitnigam, 2019-02-19:
Maybe it's worth considering a generalized `switch` proposal since `switch` will also have the same constraint.

sa2257, 2019-02-19:
Is there a discussion I can take a look at for `if` only? I fail to see why `a[0]` access unsafe only with the `else` condition. It seems to me the access will be unsafe with `if` only.

In general `switch` is a `mux` and single `if` condition is just a `mux` with one input tied down?

rachitnigam, 2019-02-19:
That's irrelevant. My point is in `if...else` only one block executes every cycle, but we don't know which one. This means we have to enforce:
(1) Each branch is allowed to consume banks independent of the other branch. For example, both branches can consume `a[0]; a[1]` and it should be fine because it is not a memory safety violation.
(2) Anything sequenced with `if...else` using `;` is scheduled to run in the same cycle. This means we need to make sure anything consumed in either branch cannot be used outside the branch.

sa2257, 2019-02-20:
Wait. So you think we should have an enable mechanism in the hardware?
That’ll be nice, but for simple structures like this, having both blocks of hardware and sampling the relevant one with a `mux` makes more sense. So we can execute both blocks every cycle, and certainly should allocate resources accordingly I think. 

What does `a[0]; a[1]` mean?

Both branches can do that because we put a `mux` in there and not because only one is executing. 

2) Yes. But just to be clear, what’s sequencing with `if else` mean?

rachitnigam, 2019-02-20:
The accepted semantics for this are to be the ones outlined in https://github.com/cucapra/seashell/issues/76#issuecomment-465355311

sampsyo, 2019-02-21:
I realized I missed this question earlier:

> @sampsyo is there a well known conservative estimations of this in linear type systems?

I think your comment demonstrates a standard-ish way to treat something like `if` in a linear (or rather, affine) way: the rule needs to type-check both branches of the `if`. To do so, the rule divides the original context (from before the `if`) into two disjoint subsets—call them Gamma1 and Gamma2. It typechecks _both_ branches of the `if` with Gamma1. Then the resulting context is Gamma2. In an affine (as opposed to linear) system, either branch can ignore part of Gamma1.

PR #81 (2019-02-21): Implement else branches
Fixes #76.

The type checker implements mostly we talked about in the meeting. The one new behavior is that both branches are required to use expressions that acquire capabilities in the same way. One branch cannot write into an array access while the other tries to read from it.

Also, this PR restructures `Environment` to better hide implementation details and even adds comments for it's methods!
closed 2019-02-21 by rachitnigam