cucapra/dahlia #79: Loop iterator type

tedbauer, 2019-02-20:
I ran into a situation where I wanted to pass a loop iterator to a function, like this:
```
def test(a: bit<32>) {
	let test2 = a;
}

for (let i = 0..5) {
	test(i);
}
```
But the typechecker is seeing `i` as type `idx<0..1, 0..5>`, and gives this error: 
``` [Type error] [6.7] Expected subtype of int32 in parameter, received: idx((0,1), (0,5)). ```
Should this fail like this?

sampsyo, 2019-02-20:
This is related to #73, FWIW!

rachitnigam, 2019-02-20:
Gah, subtyping and join need to be updated

tedbauer, 2019-02-20:
Another spot where this problem came up: for `fft`, I wanted to OR together an iterator and a `bit<32>` variable, and I got a similar error.

rachitnigam, 2019-02-20:
Paste the program here please.

tedbauer, 2019-02-20:
The original, from [here](https://raw.githubusercontent.com/breagen/MachSuite/master/fft/strided/fft.c):
```
    ...

    outer:for(span=FFT_SIZE>>1; span; span>>=1, log++){
        inner:for(odd=span; odd<FFT_SIZE; odd++){
            odd |= span;
            even = odd ^ span;

            temp = real[even] + real[odd];
            real[odd] = real[even] - real[odd];
            real[even] = temp;

            temp = img[even] + img[odd];
            img[odd] = img[even] - img[odd];
            img[even] = temp;

            rootindex = (even<<log) & (FFT_SIZE - 1);
            if(rootindex){
                temp = real_twid[rootindex] * real[odd] -
                    img_twid[rootindex]  * img[odd];
                img[odd] = real_twid[rootindex]*img[odd] +
                    img_twid[rootindex]*real[odd];
                real[odd] = temp;
            }
        }
    }
}
```

rachitnigam, 2019-02-20:
The type of `odd` and `even` will be dynamic since performing an operation on an index type upcasts it to a dynamic type. It's unclear if you should actually be allowed to do this. Instead, think about the view this implies.

Open a new issue if you think this particular operations should be applied.

commit e3ba58d5 (2019-02-20): index types can be upcast to sized ints; fixes #79
closed 2019-02-20 by rachitnigam