cucapra/dahlia #80: Alternative serial syntax proposal

tedbauer, 2019-02-20:
I think that the `---` syntax is hard to follow whenever it's used really frequently. Here's an example with reading and writing from an array serially:
```
real_even := real[even];
---
real_odd  := real[odd];
---
temp := real_even + real_odd
---
real[odd] := real_even - real_odd;
---
real[even] := temp;
```
What about supporting an additional syntax that's something like this, that makes there less need for so much whitespace?:
```
serial:
    | real_even := real[even];
    | real_odd  := real[odd];
    | temp := real_even + real_odd
    | real[odd] := real_even - real_odd;
    | real[even] := temp;
```

Or maybe it would be cool if you could write something that just means "do all of this serially", so the above could just be written like this:
```
serial: temp := real[even] + real[odd];
---
serial: real[odd] := real[even] + real[odd];
```
(This example is from [fft](https://github.com/breagen/MachSuite/blob/master/fft/strided/fft.c) in MachSuite.)

rachitnigam, 2019-02-20:
It's unclear if we want people to actually use serial composition that often. If you're trying to get parallelism out of a programs, you should have to minimize the use of `---` and `flex` views.

rachitnigam, 2019-02-28:
We're going to keep the original syntax for now.

closed 2019-02-28 by rachitnigam