cucapra/dahlia #82: Software testing harness

rachitnigam, 2019-02-21:
While we're waiting for #68, I can implement a software  harness to validate that we are implementing the right functions and emitting valid C.

Implementation Sketch
----
- Write a C program (data parser) that parses a file that contains one value on each line.
- The data parser passes an array containing all the data into a function called `harness(int* data)`.
- Each Fuse program will define a harness function that will realign the data as needed. 
  - Note: We probably need explicit casting for this.
- Compile benchmark with the data parser (using gcc) and run it on a supplied data file.
- Add a parameter to the CLI to turn all sized ints into either `double` or `int`. When compiling for the harness, use this parameter.

sampsyo, 2019-02-21:
Sounds good. But rather than going through the complexity of packing and unpacking all the data through a single flat chunk of memory (for which I’d recommend `void*` or `char*`, BTW---these are the types that the C standard says are safe to point to arbitrary data types), it might be easier to just have a per-benchmark harness program. They could all use a shared header file to simplify parsing, for example, but I think this might be less of a headache than trying to write "one harness to rule them all."

rachitnigam, 2019-02-21:
Yeah, the proposal says that each program will define a harness function. I don't know of any better ways of doing parsing and data alignment in C. Any suggestions?

sampsyo, 2019-02-21:
I’m recommending that the harness function for each app need not have the same type. The per-app harness would implement an app-specific parser (using scanf, presumably) to read an app-specific number and type of input values.

sa2257, 2019-02-28:
Hi, just wondering whether we could use the same harness to run on software too. My current setup is,
host cpp
header file
seashell hardware function
- for board testing
harness simply bundles everything and send to Buildbot, which generates HLS hardware function cpp
run `sds++` for selected configuration

**1st proposal for software testing**
Use same generated HLS hardware function cpp generated in Buildbot.
Leverage defines used for Vivado testing not to use sds libraries.

**2nd proposal for software testing**
For Vivado testing I run software and Vivado on the server. We can do the same for Seashell tests too. Two issues I have,
- adding the header file to the generated cpp seems a little hacky
- need a consistent way to run `./fuse` from scripts

sampsyo, 2019-02-28:
Yeah, the first method sounds the most straightforward to me.

PR #91 (2019-03-06): Software harness
Fixes #82.

~WIP.~

This PR defines a new backend and a C++ parsing harness that will allow us to run kernels with arbitrary data.
closed 2019-03-06 by rachitnigam