cucapra/dahlia #89: Using simple views

sa2257, 2019-02-28:
Following is my implementation for stencil
```
    for (let r = 0..22) {
        for (let c = 0..22) {
            view v_orig = shrink orig[3: r][3: c];
            for (let k1 = 0..3) unroll 3 {
                for (let k2 = 0..3) unroll 3 {
                    let mul = filter[k1][k2] * v_orig[k1][k2];
                } combine {
                    sol[r][c] += mul;
                }
            }
        }
    }
```

I get following error
```
[Error] [5.42] failure: string matching regex '(-)?[0-9]+' expected but 'r' found
```

I guess I'm confused with the syntax here.

My understanding of views,
```
view v_a = shrink a[width: offset: step]
```
width being a constant, offset can be a variable.

sa2257, 2019-02-28:
Thanks for the clarification @rachitnigam 

The syntax is `view v_a = shrink a[step * off : width]`

For shrink width = step

But for a simple iteration pattern we'd need the step to equal 1 right?

My understanding of,
width- size of view
offset- offset from original array 0th element to view's 0th element
step- step in original array between adjacent elements in the view

rachitnigam, 2019-02-28:
Yeah, shrink views don’t support step 1 iteration. We need flex views (unimplemented) for those.

sa2257, 2019-02-28:
I though shrink views are views with each element coming from a distinct bank, and one element coming from all banks in the original array. To which filtering would surely qualify. I'm a little confused by the step = width constraint.
Can you elaborate what shrink views are?

sampsyo, 2019-02-28:
The simple description of shrink views is that they just *reduce the banking factor* of the underlying array, which lets you (for example) iterate over a 10-banked array with a 5-unrolled loop.

rachitnigam, 2019-02-28:
Resolving since there is nothing to fix.

closed 2019-02-28 by rachitnigam

sa2257, 2019-02-28:
I see. So the aim of shrink views are to reason about bank factor and unroll factor mismatches?

sampsyo, 2019-03-01:
Yes, and to make manifest the additional muxing overhead that is required (relative to accessing the 10-banked array from a 10-unrolled loop).