cucapra/dahlia #101: Operations between floats and ints

tedbauer, 2019-03-01:
Is there any way to write something like this?
```
let x = 2.5;
let y = 1 + x;
```
This currently outputs:
```
[Type error] [2.11] + expected integers or floats, received: static(1) and float.
let z = 1 + x;
          ^
```

rachitnigam, 2019-03-01:
Why not turn `1` into `1.0`? It doesn't make sense to add ints and floats.

I guess a different example where this might be needed is:
```
decl x: bit<32>
decl f: float
x + f
```

sampsyo, 2019-03-01:
Yeah, it does seem like it will be handy (although not exactly research-y) to have lossless int->float conversions and rounding float->int conversions. Even making those conversions explicit, [OCaml style](https://caml.inria.fr/pub/docs/manual-ocaml/libref/Pervasives.html#VALfloat_of_int), would be fine.

rachitnigam, 2019-03-01:
Should we consider explicit casts?
```
let f = 10.0
(f as bit<32>) + 10
```

Would that break numeric subtyping?

sampsyo, 2019-03-01:
Explicit casts seem good, and they're unlikely to break subtyping (whereas implicit conversions could get confusing when they interact with subtyping).

PR #188 (2019-05-02): Explicit casting
Fixes #101
commit 0cf88a45 (2019-05-02): add support for explicit casting. Fixes #101
closed 2019-05-02 by rachitnigam