cucapra/dahlia #105: Low Priority Issue Tracker

tedbauer, 2019-03-01:
This issue tracks low-priority compiler issues that aren't important enough to warrant their own issue.

## Tasks

- [ ] 1. Semicolon parsing
- [ ] 2. Program structure error messages

### 1. Semicolon Parsing
The compiler parses loops/conditionals like this:
```
for (..) {
    ...
};
let x = 2;
...
```
It would be nice to not require a `;` after the`}` closing the loop body.

### 2. Program structure error messages
This program:
```
let x = 2;
decl test: bit<32>[5];
```
Gives this error message:
```
[Error] [2.6] failure: '---' expected but 't' found

decl test: bit<32>[5];
        ^
```
The error is due to the fact that `let` defs cannot occur above `decls` as Fuse program structure dictates, but this error message does not communicate that.

sa2257, 2019-03-03:
3. Assignment operator errors

```
for (let i = 0..1024) unroll 32 {
    c[i] = a[i] + b[i];
}
```

The error I get is 
```
[31mError[0m] [4.10] failure: '}' expected but '=' found

    c[i] = a[i] + b[i];
         ^
```
Would be nice to have an error that points to reassignment.

tedbauer, 2019-03-14:
4. No file error message

If you supply a nonexistent file to `fuse`, it throws some uncaught exception, would be nice if it had a prettier error message

tedbauer, 2019-03-14:
## 5. Record syntax
The docs are inconsistent with  #129 with record syntax

rachitnigam, 2019-03-21:
6. Fixed sized arithmetic support for C++ backend using [ac_int](https://hlslibs.org/#collapseACDatatypes).

tedbauer, 2019-04-19:
# 7. Add support for `double` type (?)
The type signatures in the header files for most of the MachSuite apps have `double` types, so having some type in Fuse that would emit a `double` in the C++ backend would be needed to write Fuse files that match the headers.

sampsyo, 2019-07-15:
N.B. item 7 above (double support) has landed!

# 8. Character literals

*Maybe* support literals like `'a'`? Their type would be `ubit<8>`, probably. Originally discussed in #211.

rachitnigam, 2020-08-12:
Stale and a bunch of these things are fixed.

closed 2020-08-12 by rachitnigam