cucapra/dahlia #96: Serial operator and variable bindings in a loop

tedbauer, 2019-03-01:
This code compiles fine, unless the two commented lines are uncommented:
```
def sum_scan(bucket: bit<32>[10], sum: bit<32>[5]) {
  let bucket_idx = 0;
  let sum_tmp = 0;
  //sum[0] := 0;
  //---
  for (let radix_id = 1..10) {
    bucket_idx := radix_id * 10 - 1;
    sum_tmp := sum[radix_id - 1];
    ---
    sum[radix_id] := sum_tmp + bucket[bucket_idx];
  }
}
```
If I uncomment those two lines, I get this error:
```
[Type error] [23.5] Variable bucket_idx not defined.
    bucket_idx := radix_id * 10 - 1;
    ^
```
The `---` operator seems to make the code following it forget the definition of `bucket_idx`. Is this intentional?

tedbauer, 2019-03-01:
This will address an issue in #96.

rachitnigam, 2019-03-03:
This is because of #51. This smaller example also fails:
```
let bucket_idx = 0;
---
bucket_idx := 10
```

Let's move the discussions to #51.

closed 2019-03-03 by rachitnigam