cucapra/dahlia #98: Macros for static constants

tedbauer, 2019-03-01:
In a lot of the MachSuite apps, there are static constants. I think it would be good to have some such feature, as I end up hardcoding a lot of magic numbers.

tedbauer, 2019-03-01:
One thing I forgot to consider: there would need to be some way to enforce that a variable created with a `const` def would be immutable.

tedbauer, 2019-03-01:
Another thing: how to preserve the type information, e.g. in the `DENSITY` example, how to make sure   the `bit<32>` information is not lost.

sampsyo, 2019-03-01:
This does indeed seem useful. For the syntax, I suggest making these exactly the same as `let`, except with `const` in there instead:

```
const DENSITY2: bit<32> = DENSITY + 5;
```

This would be the same as `let` and `const` in modern JavaScript, for example.

This is longer term, but it would be worth considering which of these constants should also be able to affect type-level parameters like array sizes and loop unrolling factors...

rachitnigam, 2019-03-03:
An alternative proposal @sampsyo and I talked about is adding type-level natural numbers:
```
type M: Nat = 10;
type N: Nat = 5;
decl a: bit<10>[M bank N];
```

The benefit of making this type level is that this can easily be extended for the implementation of #69 where we'll have a notion of type-level bounds and type-level constants:
```
type B: Bound = ?? % 4
def foo(a: bit<32>[10 bank B])
```
which will imply that the banking factor is a factor of `4`.

sampsyo, 2019-03-03:
One consequence of that route would be that we'd need a way to "decay" type-level numbers to term-level constant numbers to meet the `DENSITY` use case above.

rachitnigam, 2019-03-03:
**TLDR**:
- A `const` at the language level (not the type level) seems like feature creep.
- A complicated type-level solution allowing for "decaying" would make polymorphism really hard.
  - Any solution that brings language level constructs to type level might have this problem.

Hm, that sounds non-trivial to do. I think `const` and `type Nat` should be kept separate. Adding `const` seems to re-introduce the complicated type-level computation that we used to do in this case:

```
let x = 1;
let y = 2;
let z = x + y; // z is static(3)
```

I would push against that to avoid feature creep. As a first pass, I think polymorphism shouldn't allow computation on type level stuff. Consider this program:
```
type M : Nat = ??
type N : Nat = M * M + 1;
type K : Nat = ??
def foo(a: bit<32>[N bank K]) ...
```
which would force the solver to create the non linear constraint (it is non-linear, right?): `(= (% (M * M + 1) K) 0)`.

A counterpoint might be that a separate `const` would solve this issue, but I don't want to create several binders each with slightly different semantics.

sampsyo, 2019-03-03:
Yep, this is deceptively tricky! It seems like there are two "local minima" on opposite ends of the complexity spectrum:

1. A simple `const` modifier that doesn't grant any "special powers" like `static(n)` does—it just prevents mutation. Like `const` in JavaScript or `val` in Scala and Swift. These could of course not appear in types (e.g., they couldn't affect banking or loop unrolling factors, and they probably couldn't even be used for array sizes).
2. Type-level values. I'd argue that we would want to allow _some_ amount of type-level computation for crafting meaningful constraints, but you're right that we would want to be careful about going totally overboard. If we're not careful, we'll end up reimplementing [Liquid Haskell](https://ucsd-progsys.github.io/liquidhaskell-blog/). But in any case, these would be more flexible both in the computation that they can be involved in and the contexts they can be used in.

rachitnigam, 2019-03-03:
Right, but if that is the semantics of `const`, i.e., it doesn't give a TStatic, it won't be useful for anything @tedbauer wants it for since any operations on it would still lead to a dynamic int.

> I'd argue that we would want to allow some amount of type-level computation for crafting meaningful constraints ...
Yup, I agree. Some variant of "linear equations over bound types" might work out fine.

rachitnigam, 2019-03-12:
@tedbauer can we merge this with #69. The type level consts can be defined as:

```
type M: Nat = 1
```

The first pass of this will not allow for operations over these as a conservative restriction but we can add more features to as we implement polymorphism,

tedbauer, 2019-03-12:
Sounds good!

closed 2019-03-12 by tedbauer