cucapra/dahlia #106: Using combine operators and serial operators

sa2257, 2019-03-02:
I'm attempting to recreate this [code](https://github.com/cucapra/epoisses/blob/master/mach-gemm_ncubed/gemm.cpp)

```
for(i=0;i<row_size;i++) {
        for(j=0;j<col_size;j++) {
           
            i_col = i * col_size;
            float sum = 0;
            
            for(k=0;k<row_size;k++) {
                k_col = k * col_size;
                mult = m1[i_col + k] * m2[k_col + j];
                sum += mult;
            }
            
            prod[i_col + j]  = sum;
        }
}
```

A functionally correct translation is,
```
for (let i = 0..32) unroll 1 {
        for (let j = 0..32) unroll 1 {
            
            for( let k = 0..32) unroll 32 {
                let mult = m1[i][k] * m2[k][j];
            } combine {
                prod[i][j]  += mult;
            }

        }
}
```

However, `combine` operator assumes some sequencing within the combine scope.
i.e.
```
let x = prod[i][j]
---
prod[i][j]= mult + x
```
which we don't translate to C. 

Also, the C version is not reading `prod[i][j]`

Therefore, an alternate translation is,

```
for (let i = 0..32) unroll 1 {
        for (let j = 0..32) unroll 1 {
            let sum = 0.0;

            for( let k = 0..32) unroll 32 {
                let mult = m1[i][k] * m2[k][j];
            } combine {
                sum  += mult;
            }

            prod[i][j] = sum;
        }
    }
```

This throws an error `failure: '}' expected but 'p' found`

I assume I don't need serialize as I'm doing only one memory access, and sum is just a wire.

rachitnigam, 2019-03-02:
You’re missing a semicolon at the end of the for loop }. It needs to be };. this is a parser big.

The final translation looks fine to me functionally.

> However, combine operator assumes some sequencing within the combine scope.

If I’m understanding this correctly, yes it does a need for sequencing but we can be assured that prod[i][j] is not being read by any other concurrent hardware because combine blocks have forced serial execution.

One subtly that would make a reduction onto prod would be a data dependence created in the parallel part of the for loop.

sampsyo, 2019-03-02:
> You’re missing a semicolon at the end of the for loop }. It needs to be };. this is a parser big.

See also #105.

rachitnigam, 2019-03-02:
@sa2257 if this answers your questions, feel free to close it.

sa2257, 2019-03-02:
I just tested it!

closed 2019-03-02 by sa2257