cucapra/dahlia #116: Seashell generated C code has pointers not arrays

sa2257, 2019-03-10:
Previously generated [cpp](http://gorgonzola.cs.cornell.edu:8000/jobs/GHxxX5ed_8Q/files/code/gemm.cpp) for  `gemm`  example running on Buildbot

Currently failing [cpp](http://gorgonzola.cs.cornell.edu:8000/jobs/4htdM6yWs7c/files/code/gemm.cpp) for `gemm` example

rachitnigam, 2019-03-10:
Hm, I don't have access to gorgonzola from home. Can you paste the build logs here/tell me how to access gorgonzola from non-Cornell network?

tedbauer, 2019-03-10:
I don't know if this would help, but there is a Cornell VPN: https://it.cornell.edu/cuvpn

rachitnigam, 2019-03-10:
Works! Thanks.

sa2257, 2019-03-10:
If it helps, 
previous(from branch builbbot-hls-simplify)
```
void gemm(float m1[4096], float m2[4096], float prod[4096]) {
  #pragma HLS ARRAY_PARTITION variable=m1 factor=64
  #pragma HLS ARRAY_PARTITION variable=m2 factor=64
  
  for(int i = 0; i < 64; i++) {
    
    for(int j = 0; j < 64; j++) {
      
      float sum = 0.0;
      for(int k = 0; k < 64; k++) {
        #pragma HLS UNROLL factor=64
        float mult = (m1[(k + (64 * i))] * m2[(j + (64 * k))]);
        // combiner:
        sum += mult;
      }
      prod[(j + (64 * i))] = sum;
      // combiner:
      
    }
    // combiner:
    
  }
}
void kernel() {
  
  
}
```

current, (local branch merged with integration-data_extraction and builbbot-hls-simplify)
```
void gemm(float *m1, float *m2, float *prod) {
  #pragma HLS ARRAY_PARTITION variable=m1 factor=64
  #pragma HLS ARRAY_PARTITION variable=m2 factor=64
  for(int i = 0; i < 64; i++) {
    
    for(int j = 0; j < 64; j++) {
      
      float sum = 0.0;
      for(int k = 0; k < 64; k++) {
        #pragma HLS UNROLL factor=64
        float mult = (m1[(i + (64 * k))] * m2[(k + (64 * j))]);
        // combiner:
        sum += mult;
      }
      prod[(i + (64 * j))] = sum;
      
    }
    
  }
}
void kernel() {
  
  
}
```

rachitnigam, 2019-03-10:
Is there a way to see the build log? I want to see the error that buidbot shows.

sa2257, 2019-03-10:
Can you clarify why the change was intentional?

rachitnigam, 2019-03-10:
Slightly easier to emit, but I can revert it back easily.

sa2257, 2019-03-10:
Yes, 
[prior](http://gorgonzola.cs.cornell.edu:8000/jobs/GHxxX5ed_8Q.html)
[now](http://gorgonzola.cs.cornell.edu:8000/jobs/4htdM6yWs7c.html)

rachitnigam, 2019-03-10:
Offending error:
```
ERROR: [SYNCHK 200-61] /seashell/buildbot/instance/jobs/4htdM6yWs7c/code/gemm.cpp:11: unsupported memory access on variable 'm1' which is (or contains) an array with unknown size at compile time.
INFO: [SYNCHK 200-10] 1 error(s), 0 warning(s).
ERROR: [HLS 200-70] Synthesizability check failed.
```

sa2257, 2019-03-10:
I see, HLS requires array size. One way to do it is through array arguments. Other way would be through pragmas in header file. I think giving it through arguments is cleaner,
1. Hardware function actually reads in an array not a pointer
2. Input output sizes are seen from hardware function itself

sa2257, 2019-03-10:
Yes, I traced back from the error.

rachitnigam, 2019-03-10:
Try the fix in https://github.com/cucapra/seashell/pull/117 and let me know if it works.

commit f2acfcb1 (2019-03-10): fixes #116
commit 0e6c3cc7 (2019-03-11): fixes #116 (#117)

* fixes #116

* emit flattened length
closed 2019-03-11 by rachitnigam