cucapra/dahlia #123: Int type inference behavior

tedbauer, 2019-03-13:
I'm a little unclear if #111 subsumes this, but right now, when this program compiles just fine:
```
decl arr: bit<32>[10];

for (let i = 0..33) {
  arr[5] := i;
}
```
However, this program:
```
decl arr: bit<32>[10];

for (let i = 0..33) {
  arr[5] := i * 1;
}
```
Throws this error:
```
Exception in thread "main" java.lang.RuntimeException: [Type error] [4.13] Expected subtype of int32 in assignment, received: int33.
  arr[5] := i * 1;
            ^
```
Is this the expected behavior?

rachitnigam, 2019-03-13:
Try using an explicit let binding to get the types to work? See if that helps?

tedbauer, 2019-03-13:
This program compiled successfully:
```
decl arr: bit<32>[10];

for (let i = 0..33) {
  let temp:bit<32> = 1;
  arr[5] := i * temp;
}
```
So there are ways to write this so it'll type check, but I wasn't sure if the error message from my first example was erroneous. Should the type be `bit<33>` like it says?

rachitnigam, 2019-03-14:
This is a bug. Here is the offending code: https://github.com/cucapra/seashell/blob/f65229d3997af8e69974954da5daa8fb38207202/src/main/scala/Subtyping.scala#L78

commit c5418cf4 (2019-03-14): fixes #123
closed 2019-03-14 by rachitnigam