cucapra/dahlia #124: Proposal: New view combinators

sampsyo, 2019-03-14:
As we discussed today, and as a successor to #57's "shrink" and "flex," let's consider a new set of views. I'm calling these "combinators" because they are ways to take a memory and build a new memory out of it. (The output memory is, from the language's perspective, indistinguishable from a real, physical memory, so it can also be used in views.)

The goal is support the same patterns as in #57. As a quick summary, we're focused on convolution-style moving-window iteration (with overlap) and JPEG-style blocked iteration (without overlap). In the former case, we want to be able to unroll the inner loop (over the filter). In the latter case, we want to be able to unroll both loops.

So, let's build up a list of sufficient view combinators that get us there. For each combinator, we need four things: (1) the semantics, i.e., what values from the old memory show up in the new memory; (2) the banks to consume of the old memory; (3) the type of the new memory; and (4) some notion of the hardware cost.

Here's the list from today:

1. **Suffix**, a.k.a. cheap suffix. A slice expression like `a[i:]` in Python, where `i` must be a multiple of `a`'s banking factor. Consumes all of `a`. Produces a new, smaller memory with the same banking factor. The hardware is a cheap address offset.
2. **Rotation suffix**, a.k.a. expensive suffix. Also a slice like `a[i:]` in Python, but `i` is unconstrained. Also consumes all of `a` and produces a memory with the same banking factor. The hardware is a more costly bank rotation circuit.
3. **Prefix**. A slice like `a[:i]`. Consumes all of `a` and produces a smaller memory with the same banking factor. There is no hardware cost; this is just changes the type of the memory to allow access to a smaller part of it. For what it's worth, this combinator seems to argue in favor of *sound* bounds checking… otherwise, it would be possible to access the memory past the end of a prefix.
4. **Split**. Divide the banks of a memory into multiple, smaller memories. The semantics are kind of hard to explain, but the idea is that each of the several resulting memories represent discontiguous (strided) chunks of the original underlying array. The *total* banking factor of the resulting memories equals the banking factor of the original array, i.e., banks are neither created nor destroyed. All the banks of the original memory are consumed. The implied hardware seems to consist of almost nothing at all.
5. **Shrink**. Reduce the banking factor of an underlying array by an integer factor. The output memory is the same size as the input memory, just with a smaller banking factor—and the entire original array is consumed. The implied hardware is just enough fairly cheap muxing to allow N parallel PEs to access k*N banks by switching between bank sets in each iteration group.

We're getting closer! Some careful thought is still required to show precisely how to use these combinators to make our two notional loops we care about (convolution-like and JPEG-like) type-check.

sa2257, 2019-03-14:
> ... they (combinators) are ways to take a memory and build a new memory out of it. (The output memory is, from the language's perspective, indistinguishable from a real, physical memory, so it can also be used in views.)

To state the obvious, the combinator will be a restricted subset of 'the' physical memory within that scope, isn't it?

> 3. **Prefix**. A slice like `a[:i]`. Consumes all of `a` and produces a smaller memory with the same banking factor. There is no hardware cost; this is just changes the type of the memory to allow [...]

Is it useful to do a check with the corresponding suffix too? Ie. whether prefix-suffix is divided by banking factor or not, which if not has some hardware cost associated with it?

sa2257, 2019-03-14:
I'm still struggling to clearly see what "split" is.

> 4. **Split**. Divide the banks of a memory into multiple, smaller memories. The semantics are kind of hard to explain, but the idea is that each of the several resulting memories represent [...]

My current understanding is it's an alternate way to express multi-dimensions?

JPEG like access?
```
in: float[8 bank 2][4 bank 4]
out: float[8 bank 2][4 bank 4]

    for (let i = 0..8) unroll 2 {
        for (let j = 0..4) unroll 4 {
            out[i][j] :=  f(a[i][j]);
        }
    }
```
Am I missing something in this example?

Vivado HLS would require you to use `ARRAY PARTITION cyclic` on two dimensions to match this pattern. This corresponds to `ARRAY PARTITION cyclic factor 8` on one dimension, because inner loop is completely unrolled.

I'm imagining an alternative way of expressing this partitioning on a single d array would be (I don't think this is permitted in HLS),
`ARRAY PARTITION block factor 4, cyclic factor 4` which will be useful if one doesn't want to completely unroll the inner loop, ie.`ARRAY PARTITION block factor 4, cyclic factor 2`, which is inexpressible with just cyclic factors in one d, but simple affair in multi d.

So, 
1. Is 'split' trying to address something beyond multi d arrays?
2. If so, does it contain something beyond partitioning patterns?

sampsyo, 2019-03-14:
> To state the obvious, the combinator will be a restricted subset of 'the' physical memory within that scope, isn't it?

Yes.

> Is it useful to do a check with the corresponding suffix too? Ie. whether prefix-suffix is divided by banking factor or not, which if not has some hardware cost associated with it?

What do you mean by "prefix-suffix is divided by banking factor"? What is the hardware cost you are imagining?

> My current understanding is it's an alternate way to express multi-dimensions?

There is some similarity here to two-dimensional array banking, but the point here is that we want to be able to process data that is *logically one-dimensional* in this style. Or data that is *logically two-dimensional*, but we want to unroll the loops that iterate over both dimensions.

> 1. Is 'split' trying to address something beyond multi d arrays?
> 2. If so, does it contain something beyond partitioning patterns?

The idea is to be able to access one-dimensional array data in a doubly-unrolled way. So that goal precludes changing the dimensionality of the *original* array. I'm still not quite sure what you mean by "partitioning patterns"—we're still just talking about a cyclically partitioned underlying array. Can you explain why the array partitioning schemes you've laid out above are useful for this case? I still don't see it.

sa2257, 2019-03-14:
Rotate access with the new proposal?
Unsure how to specify accessor to rotate combinator
```
in: float[9 bank 3]
f: float[3 bank 3]
out: float[7]

    c_in = in[0:8]<rotate> // associate k' and unroll 3
    for (let i=0..7){
        for (let k = 0..3) unroll 3 {
            let temp = f[k] * c_in[k']; // k' is i+k
            } combine {
                out[i] += temp;
            }
        }
    }
```

sa2257, 2019-03-14:
> What do you mean by "prefix-suffix is divided by banking factor"? What is the hardware cost you are imagining?
> 

Simply, the cost associated with skipping over exhausted banks. 

from the example `a: [7 bank 5]` from yesterday's discussion, we currently don't allow this configuration with banking. So a corresponding check in combinators on prefix - suffix.

The hardware cost can be a mux on some unrolled PEs to skip some iterations, and mux on memory accessor to skip invalid accesses.

sa2257, 2019-03-14:
> There is some similarity here to two-dimensional array banking, but the point here is that we want to be able to process data that is _logically one-dimensional_ in this style. Or data that is [...]

Maybe, I'm looking at this with an implicit assumption that _logical two-dimensional_ in Seashell world mean we intend to access this array in two dimensions, and the memory is laid out to support that.

It makes the designer explicitly express the memory layout to support their computation, (unrolling in two dimensions require a two dimensional memory), and Seashell will translate this layout to an efficient underlying one dimensional array.

Maybe there's an additional use case for multi dimensions we are hoping to explore here?

sa2257, 2019-03-14:
> The idea is to be able to access one-dimensional array data in a doubly-unrolled way. So that goal precludes changing the dimensionality of the _original_ array. I'm still not quite sure what you [...]

Okay, I see what you mean now! I agree why this has nothing to do with blocks.

But is there a use case for doubly unrolling single d arrays? It seems split is far more restrictive, and covers only one pattern that multi-d covers. Is changing the dimensionality of the original array too restrictive?

One possibility I can think of is different access patterns at different points of the design. But this is given,
- original array is cyclically partitioned (reasonable- we only permit this)
- inner loop is fully unrolled (unclear why that's reasonable)

So the use case seems require applications with
- different access patterns during the design
- these access patterns either single unrolled or doubly unrolled with inner loop fully unrolled

I imagine JPEG to not strongly require single unrolled, which permits the designer to create a 2D original array to express memory layout for the computation intended?

sa2257, 2019-03-14:
Trying to motivate an example:

case with no overlap (no rotation)
```
a: float[4 bank 4] // [2 bank 2][2 bank 2]
c: float[2 bank 2]
f: float[2 bank 2]

    s_a = split(a,2)
    for (let i = 0..2) unroll 2 {
        for (let j = 0..2) unroll 2 {
            //c[i] :=  a[i][j]*k[j];
            c[i] :=  s_a[i][j]*k[j];
        }
    }
// seems straight-forward to imagine 'a' as a 2D array
```

case with overlap (rotation) (unroll choice and order matters)
this is also an instance where array dimension and unroll factors don't correspond well (possible use case?)
```
a: float[4 bank 4] // Only have one dimension
c: float[3 bank 3]
f: float[2 bank 2]

    s_a = split(a,2)
    for (let i = 0..3) unroll .. {
        for (let j = 0..2) unroll 2 {
            c[i] :=  s_a[i][j]*k[j];
        }
    }
// unroll factors of j and k seems to be dependent on each other. Reflects underlying '1Dness' of 'a'?
```

sampsyo, 2019-03-15:
> Simply, the cost associated with skipping over exhausted banks. [...] The hardware cost can be a mux on some unrolled PEs to skip some iterations, and mux on memory accessor to skip invalid accesses.

Aha! If possible, I would love to think of this—i.e., an array size that is not a multiple of the banking factor, requiring some manner of masking at the end of unrolled iteration—as a separate issue: that is, an issue that applies to physical memories just as much as it does to views. We probably even need a separate GitHub issue for this. Whatever mechanism we invent there (for physical memories) will also apply here (for views).

> But is there a use case for doubly unrolling single d arrays?

The issue here, as I mentioned above, is that the data is *logically one-dimensional*. So the programmer wants to think of it as a 1D array, load it in from a file as a 1D array, print it out as a 1D array, write comments about the 1D accesses, etc. There's an analogy here to a (much) earlier conversation about *multidimensional* arrays. We have decided to have multidimensional arrays in the language, even though we *could* have just had 1D arrays and indexed them in a `w*i+j` style, but we decided we wanted n-dimensional arrays because that's how the programmer wants to think of them—as *logically multidimensional*. In the same way, I would like not to add spurious dimensions to arrays just to admit a certain iteration pattern—I would like the shape of an array to reflect its logical structure.

rachitnigam, 2019-04-05:
Syntax forms
--------------

- `ve ::= a[suf:<npre> bank <nshrink>]`: Sugar for combining `prefix`, `suffix`, and `shrink` views easily. Combining all these forms together is reasonably because `prefix` and `shrink` don't require any changes in code generation. Only `suffix` affects the code generation.
- `suf ::= n | n!`: `n!` implies unconstrained rotation.
- `ve ::= split[2](a)`: Split view over `ve`. (**I'm not particularly fond of this syntax**)
- `cmd ::= ... | view (v1, v2 ...) = ve2`: Top level commands to create views. These could've been simple `let` bindings but that requires add views to the syntax category of `e`.

rachitnigam, 2019-04-05:
On Multidimensional Splits
-----------

Two observations about (multidimensional) splits:
- Splitting does not reduce the number of dimensions in the resulting views. If `a[_ bank 4][_ bank 4]` is split by `a[split 2][split 2]`, it creates 4 views, each of which are 2-dimensional `v[_ bank 2][_ bank 2]`.
- Splits are a *very* low-level primitive. It seems the kinds of iterations one might actually want to do after a split requires tons of code duplication:
```
decl A: int[M bank 4][N bank 4];

view ((v1, v2), (v3, v4)) = a[_ split 2][_ split 2]; 
// vi: int[M/(4/2) bank 2][N/(4/2) bank 2]

for(...) {
  let v = v1[i]
} combine {
  B[0] += v
};
... // 3 more copies of the same loop sequenced in parallel.
```
There is probably a more expressive version of `unroll` that needs to exist that can duplicate arbitrary patterns like these...

sampsyo, 2019-04-06:
The proposed syntax looks good—and yeah, I agree that a separate `view` command is probably what we want (instead of making views into expressions).

For `split`, I think I'm most fond of yet another separate command, like the `split (v1, v2) = a by 2` proposal. This could even be simplified to `split (v1, v2) = a` in the single-dimensional case, at least, because the number of division is implied by the LHS.

It's very true that splitting does lead to some weird code duplication. It remains somewhat unclear to me how real programs will actually want to use splits… it would be interesting to see how gnarly this becomes in practice.

rachitnigam, 2019-04-08:
Codegen for Splits
------------

I came across another issue with code gen for splits. Please carefully look over the derivation to make sure I didn't make any mistakes first:

**B_off**: Bank offset of the split array
**i**: index in the split
**B_a**: Total banks in the underlying array
**B_v**: Total banks in the split array

All variables starting with caps are compile-time constants

```
a{b}[j] = a[i]
// where
i = B_a * j + b
b = i % B_a
j = floor(i / B_a)
```

We have,
```
split (v1, v2) = a
```
Which gives us `B_off(v_1) = B_a/2 * 0` and `B_off(v_2) = B_a/2 * i`

Next, we have
```
v{b}[j] = a{b + B_off(v)}[j]
```
This gives the intuition to why we think splits have a low/no hardware overhead. However, Vivado HLS doesn't let us access the underlying banks. Thus, we need to compute `j` such that:
```
v[i] = a[j]
```
First, let's calculate `b` and `j` in this expression:
```
v[i] = a{b}[j]
// where
b = (i % B_v) + B_off(v)
j = floor(i / B_v)
```
which gives us:
```
v[i] = a[B_a * floor(i / B_v) + (i % B_v + B_off)]
```
which unfortunately seems like something that would **kill** performance. The issue is with the `/` inside the floor. If we somehow move `B_a` into the floor and rewrite `B_v` in terms of `B_a`, we could probably eliminate the `/`. Unless I'm missing something obvious about floor, this will be annoying...

rachitnigam, 2019-04-11:
@sampsyo I missed this during our conversation about `split` yesterday, but having split create a new dimension doesn't actually fix the problem above. That is, in this program:
```
a: bit<32>[8 bank 4]
split v = a[by 2]
```
The underlying structure for `a` is:

| 0 | 1 | 2 | 3 | 0 | 1 | 2 | 3 |
|---|---|---|---|---|---|---|---|

and for `v_a` is:

| 0 | 1 | 0 | 1 |
|---|---|---|---|
| 2 | 3 | 2 | 3 |

which means for the access: `v[i][j]`, we still end up producing the expression:
```
a[i * B_v/B_a + floor(j / B_v) * B_a]
```

sampsyo, 2019-04-11:
Right. The essence of the problem here is that, while we know these two layouts are _actually the same_ in terms of the contents of the underlying memory banks, there's no way to convey that to the HLS tool. Put differently, given a "fantasy" Vivado that did exactly what we wanted, we could ask for two names for the same set of four underlying banks: one single-dimensional array (`a`) and one two-dimensional array (`v`). We would know the way the indices correspond—that is, which `a[i]` corresponds to which `v[i][j]`—but the generated code could use either style to talk about the same physical data element.

Unfortunately, I bet that real Vivado doesn't let us create differently-shaped aliases like that.

Here's a very very hacky suggestion: as a concession to the HLS backend, invert the direction of the view in the generated code! Limit programs to a single `split` view per array. Then, generate code that tells Vivado that we want the higher-dimensional version instead of the (true) lower-dimensional version. Then the codegen for accessing `v` requires no index math, but accessing the original `a` requires some math. But hopefully simpler math.

Is that too horrible of an idea to contemplate? (Clearly, the real solution to this is fuse2rtl!)

rachitnigam, 2019-04-12:
Hm, I don't like the hack, but it might be our only shot. One thing to clarify here is that you are not suggesting changing the type checker to disallow multiple split views over the same array. Instead, this should be implemented as a separate well formedness check.

I haven't thought of how hard this would make having separate, non-split views on an array that is split down the line.

sampsyo, 2019-04-12:
Yep, that's exactly right.

The boil-the-ocean alternative, I suppose, would be the route where we do translate down into a core-calculus-like language that represents individual banks discretely and calculates physical addresses in each. But I think a more minimal approach would be best to ensure we make progress…

rachitnigam, 2019-04-13:
After discussion with @sampsyo, I'm implementing the dumb emissions with divisions first. I'll add support for the "smarter" codegen that creates a two dimensional array if there is exactly one `split` is used on an array.

sa2257, 2019-04-24:
@rachitnigam regarding an example for split, according to last meeting's discussion I think it's merely splitting the array into different dimensions only for computation. Isn't this similar to,

```
in: float[8 bank 4]
split v_in = in[by 2]
    for (let i = 0..2) unroll 2 { 
        for (let j = 0..2) {
            access v_in;
        }
    }
```

Why `i`?
> Which gives us `B_off(v_1) = B_a/2 * 0 and B_off(v_2) = B_a/2 * i` 

Not sure what you mean by
> This gives the intuition to why we think splits have a low/no hardware overhead. 

But `floor(i / B_a)` comes up in original array `a` access too, I don't see why this becomes an issue? 
> which unfortunately seems like something that would kill performance

PR #180 (2019-04-26): Implement Views proposal
Fixes #124
closed 2019-04-26 by rachitnigam