cucapra/dahlia #127: Loop unrolling by HLS

rachitnigam, 2019-03-14:
Reading from this page on [how Vivado generates unroll loigc](https://japan.xilinx.com/html_docs/xilinx2017_4/sdaccel_doc/uyd1504034366571.html)

Excerpts:
> To unroll a loop completely, the loop bounds must be known at compile time. This is not required for partial unrolling.

So what happens in partial `unroll`s?
> Partial loop unrolling does not require N to be an integer factor of the maximum loop iteration count. Vivado HLS adds an exit check to ensure that partially unrolled loops are functionally identical to the original loop. For example, given the following code: 
```C
for(int i = 0; i < X; i++) {
  pragma HLS unroll factor=2
  a[i] = b[i] + c[i];
}
```
> turns into:
```C
for(int i = 0; i < X; i += 2) {
  a[i] = b[i] + c[i];
  if (i+1 >= X) break;
  a[i+1] = b[i+1] + c[i+1];
}
```

Woah, didn't know that. Turns out we don't need this exit check because we statically ensure all loops are aligned with their `unroll`s. Vivado gives us control over this: 
>  However, if you know that the specified unrolling factor, 2 in this example, is an integer factor of the maximum iteration count X, the `skip_exit_check option` lets you remove the exit check and associated logic.

Concretely, we need to do two things:

1. Do we want to support misaligned looks like these? (I think no, because we can always use the pattern from #72)

2. Start emitting `skip_exit_check` option for all our unrolls.

sa2257, 2019-03-14:
I don't think we need `skip_exit_check` for simpler apps, it says,
> Because the maximum iteration count X is a variable, Vivado HLS may not be able to determine its value and so adds an exit check and control logic to partially unrolled loops.

I don't think it adds the exit check when it knows the bound. For variable bounds, we don't support them.

I'm not sure if we want misaligned `unroll`s. I'm hoping we'd run into a good example if we need it. Not sure how #72 can be used?

rachitnigam, 2019-03-14:
Can you run an experiment to validate this claim? I'm not sure how smart it is. For example, would it generate a bound for `for(...; x < 1 + 2; ...)`?

Also, what's the harm is adding it anyways?

sa2257, 2019-03-14:
Yeah, we can test it! I don't think there's any harm. HLS C is just backend generated code, probably useful to be explicit. I'm just saying I think it's not needed.

rachitnigam, 2019-03-14:
Makes sense. Verifying it would be useful and in the paper we can substantiate the claim that we can produce "better" code by showing no loops have magic guards like these.

sa2257, 2019-03-14:
The 'magic guard' is what we avoid with not allowing variable loop unroll. Being able to specify it is actually a good thing(partially) for HLS.

rachitnigam, 2019-03-14:
I disagree. HLS magically adds a guard without any errors or explanations it seems. I'm reading this has them doing a "best-effort" compilation where they add random things to the circuit.

Your point is actually maybe more valid for normal software engineering where this can just be a "trap" in the runtime which throws a nice error during runtime (so, for example why python or Java would do, not segfaulting like C/C++).

There are no nice "traps" like that in hardware from what I can tell.

sampsyo, 2019-03-17:
It does seem like it would be nice to be able to insert that `skip_exit_check` option when we can prove it's not needed (which is true of all our current loops)! Especially if we can show experimentally that this results in a hardware efficiency win of some manner.

The discussion of the new views in #124 is relevant here, however. If the prefix view combinator is unconstrained, it can be used to create memories whose size is not a multiple of the banking factor (see https://github.com/cucapra/seashell/issues/124#issuecomment-473138379). In that case, an unrolled loop over the view would require an exit check like this. That comment I linked to suggested the idea of a *separate* proposal for allowing memories whose lengths are not multiples of their banking factors—these memories (whether physical or from views) would come with the extra cost of exit checks.

PR #156 (2019-03-24): Skip exit checks in unrolled loops
Emit `skip_exit_check` for unroll pragmas. According to the manual:
    
`skip_exit_check`: An optional keyword that applies only if partial unrolling is specified with `factor=`. The elimination of the exit check is dependent on whether the loop iteration count is known or unknown:
- Fixed (known) bounds: No exit condition check is performed if the iteration count is a multiple of the factor. If the iteration count is not an integer multiple of the factor, the tool:
   - Prevents unrolling.
   - Issues a warning that the exit check must be performed to proce [... cut]
closed 2019-03-24 by rachitnigam