cucapra/dahlia #130: sshpass command not found on Buildbot logs

sa2257, 2019-03-14:
Some full flow runs I made yesterday failed with 

`
$ sshpass -p root scp -r ... zb1:/mnt
command sshpass not found
`

in [gemm](http://gorgonzola.cs.cornell.edu:8000/jobs/xzWl5cF8hvI.html) through [fft](http://gorgonzola.cs.cornell.edu:8000/jobs/_BqinSApUPQ.html)

But one job [stencil](http://gorgonzola.cs.cornell.edu:8000/jobs/4yLFvuUxgyQ.html) passed.

One possibility for this to happen is if the command is run from xilinx container. But I don't see why it should.

sampsyo, 2019-03-14:
I think that should be fixed since this change, which was only deployed today:
https://github.com/cucapra/seashell/commit/d90a09c1df25998b460ea319eab09ac5e4ea0faa

sa2257, 2019-03-14:
I see the log now, it's run manually today.

closed 2019-03-14 by sa2257