cucapra/dahlia #135: Static Bounds Checking Proposal

sgpthomas, 2019-03-16:


sampsyo, 2019-03-16:
Sounds good overall, as a first step!

> TSizedInt: Output warning if length of sized int > size of array

Specifically, I think you mean emitting a warning when the *maximum value* of the sized integer is greater than the array length. (The maximum value is 2^length - 1.)

To also note some next steps:

- We talked about giving types to operations that can put an index "in bounds." For example, if `x` has type `bit<32>`, then `x & 0xFF` should probably have the type `bit<8>`. Or we could introduce more meaningful conversion syntax, as in #111, like `x as bit<8>` to do explicit truncation. Some version of this seems like a really good idea to me. A nice upshot is that this kind of truncation is basically free in hardware (just drop some wires!).
- This is a little less obviously a win, but we could consider flow-sensitive typing to support explicit dynamic checks. The idea would be that, even if `a[x]` is illegal, then `if (x < 100) { a[x] }` would be legal as long as the size of `a` is at least 100. This would require introducing a kind of type like `TSizedInt` but with an arbitrary maximum that might not be a power of two. Then we'd add flow-sensitive typing to change the type of `x` inside the `if` branch.
- With some combination of the above, we might arrive at a situation where bounds checking would be *sound*, which would be really nice—and perhaps feasible in our domain, where it's usually too annoying in C, for example. Requiring programs to do explicit truncation before indexing doesn't seem like too onerous a trade-off to get soundness in return.

sgpthomas, 2019-03-16:
Consider something like:
```
decl a: bit<32>[10];

for (let i = 0..10) {
  a[i + 0] := 1;
}
```

Currently we cast `i` up to a TSizedInt when we add 0. Under this current proposal, we would just emit a warning here, which is a little ridiculous. It seems like we could join (<index type> + <static int>) to an index type rather than a sized int.

sampsyo, 2019-03-17:
> It seems like we could join (index type + static int) to an index type rather than a sized int.

Another strategy would be to give `i + 0` a new kind of type for dynamic integers with a known range, as I briefly suggested above:

> This would require introducing a kind of type like `TSizedInt` but with an arbitrary maximum that might not be a power of two.

rachitnigam, 2019-03-17:
>  It seems like we could join (index type + static int) to an index type rather than a sized int.

Consider this:
```
for (let i = 0..4) unroll 2 { a[i] }
```
which, when desugared, is:
```
for (let i = 0..2) {
  a{0}[i];
  a{1}[i]
}
```
on the other hand,
```
for (let i = 0..3) unroll 2 { a[i + 1] }
```
which doesn't have a simple translation into the desugared world. Note that all the accesses are still *safe*, but the (bank, index) pairs are offset in that the first iteration will access (0, 1) and (1, 0) which means the underlying hardware is more complicated than the first example.

In general, support accesses of the form `a[K*i + j]` is non trivial. This is why the language has views (#124).

One thing that's worth considering is separating the bounds checking phase from simple typechecking. You can have the type checker annotate AST nodes for you and then reuse that information in a new phase.

sgpthomas, 2019-03-17:
Ok, I see why the hardware complicates things. Maybe the thing to do is have a ranged integer type.

I've implemented basic bounds checking in the typechecker already but I agree that it would be much cleaner to have this be a separate pass. It will make it easier to think about in the future.

Do you guys use pull requests?

rachitnigam, 2019-03-17:
Yeah, pull requests and branches are the preferred way of having code discussions.

sgpthomas, 2019-03-18:
Atm, my code errors on this program (`should-compile/shrink-view-simple`). The loop should go from `0..8` right?
```
decl a: bit<10>[16 bank 8];                                                                                                                                               
decl b: bit<10>[8];                                                                                                                                                       
                                                                                                                                                                          
for (let i = 0..12) {                                                                                                                                                     
  view v_a = shrink a[2*i:2];                                                                                                                                             
  b[i] := v_a[0] + 2*v_a[1]                                                                                                                                               
}   
```

rachitnigam, 2019-03-18:
Yeah, that seems reasonable. I recommend leaving out the logic for views for now. We are going to redo them in #124.

sgpthomas, 2019-03-18:
I don't have any particular logic for views. This is failing because the array `b` only has a size of `8` but `i` ranges from `0..12`.

rachitnigam, 2019-03-18:
Oh, I see. Yeah, if the size of `b` can't support `i` in range `0..12` then it should errors. Just to be clear, when you say "your code", it's the bounds checking code and not the type checker.

sgpthomas, 2019-03-18:
yes, I mean the bounds checking code

rachitnigam, 2019-03-21:
Merged #141!

closed 2019-03-21 by rachitnigam