cucapra/dahlia #148: [Impossible] error triggered with fuse run

tedbauer, 2019-03-20:
In my `seashell/` dir, I ran `./fuse run ../fuse-benchmarks/ted/bfs.fuse -o bfs.cpp`.

Contents of `bfs.fuse`:
```
record edge_t {
  dst: bit<32>
}

record node_t {
  edge_begin: bit<32>;
  edge_end: bit<32>
}

decl nodes: node_t[50];
decl edges: edge_t[50];
decl starting_node: bit<32>;
decl level: bit<32>[50];
decl level_counts: bit<32>[50];

let cnt = 0;

// Temporary variables for serial updates
let tmp_begin:bit<32> = 0;
let tmp_end:bit<32>   = 0;
let tmp_dst:bit<32>   = 0;
let tmp_level:bit<32> = 0;

let MAX_LEVEL = 10;

let tmp_l:bit<32> = 0;

for (let horizon = 0..50) {
  cnt := 0;
  for (let n = 0..50) {
    tmp_l := level[n];
    ---
    if (tmp_l == horizon) {
      tmp_begin := nodes[n].edge_begin;
      ---
      tmp_end := nodes[n].edge_end;
      ---
      for (let e = 0..50) { // FIXME: find upper bound
        if (e >= tmp_begin && e <= tmp_end) {
          tmp_dst := edges[e].dst;
          tmp_level := level[tmp_dst];
          ---
          if (tmp_level == MAX_LEVEL) {
            level[tmp_dst] := horizon + 1;
	          ---
	          cnt := cnt + 1;
          }
        }
      }
    }
  }
}
```
The compiler emitted this error:
`[Impossible] Cannot parse type node_t[50 bank 1] with CppRunnable backend. This should never trigger. Please report this as a bug.`

rachitnigam, 2019-03-20:
Ah, that's because the C++ backend doesn't know how to read in structs as inputs. I'll fix the error message to not be impossible and implement this parsing.

rachitnigam, 2019-03-20:
This is going to take some time since the C++ parsing library needs to be rewritten in some ways.

PR #149 (2019-03-21): Parsing for records
Fixes #148.

Rewrite the parsing library to use [json.hpp](https://github.com/nlohmann/json) which provides cleaner interfaces.
- Each record def generate parsing helpers which are used by the json library to automagically convert json elements into records.
- Cleaned up some old code in CppRunnable.scala.
closed 2019-03-21 by rachitnigam