cucapra/dahlia #150: fuse run: "Header json.hpp is missing from location _headers"

tedbauer, 2019-03-22:
When I ran this:
```
./fuse run bfs.fuse -o bfs.cpp
```
The compiler emits this:
```
[WARN] A SizedInt is used for an array access! This could be unsafe.
          tmp_level := level[tmp_dst];
                       ^
[WARN] A SizedInt is used for an array access! This could be unsafe.
            level[tmp_dst] := horizon + 1;
            ^
Exception in thread "main" fuselang.Errors$HeaderMissing: Header json.hpp is missing from location _headers.
	at fuselang.GenerateExec$.$anonfun$generateExec$1(GenerateExec.scala:58)
	at fuselang.GenerateExec$.$anonfun$generateExec$1$adapted(GenerateExec.scala:56)
	at scala.collection.immutable.List.foreach(List.scala:392)
	at fuselang.GenerateExec$.generateExec(GenerateExec.scala:56)
	at fuselang.Main$.$anonfun$main$4(Main.scala:81)
	at scala.util.Either.flatMap(Either.scala:341)
	at fuselang.Main$.main(Main.scala:79)
	at fuselang.Main.main(Main.scala)
```

Here are the contents of `bfs.fuse`:
```
record edge_t {
  dst: bit<32>
}

record node_t {
  edge_begin: bit<32>;
  edge_end: bit<32>
}

decl nodes: node_t[50];
decl edges: edge_t[50];
decl starting_node: bit<32>;
decl level: bit<32>[50];
decl level_counts: bit<32>[50];

let cnt = 0;

// Temporary variables for serial updates
let tmp_begin:bit<32> = 0;
let tmp_end:bit<32>   = 0;
let tmp_dst:bit<32>   = 0;
let tmp_level:bit<32> = 0;

let MAX_LEVEL = 10;

let tmp_l:bit<32> = 0;

for (let horizon = 0..50) {
  cnt := 0;
  for (let n = 0..50) {
    tmp_l := level[n];
    ---
    if (tmp_l == horizon) {
      tmp_begin := nodes[n].edge_begin;
      ---
      tmp_end := nodes[n].edge_end;
      ---
      for (let e = 0..50) { // FIXME: find upper bound
        if (e >= tmp_begin && e <= tmp_end) {
          tmp_dst := edges[e].dst;
          tmp_level := level[tmp_dst];
          ---
          if (tmp_level == MAX_LEVEL) {
            level[tmp_dst] := horizon + 1;
	          ---
	          cnt := cnt + 1;
          }
        }
      }
    }
  }
}
```

rachitnigam, 2019-03-22:
Remove _headers from the current folder, run abt assembly. Unfortunately when a new header is added, _headers needs to be completely deleted to force an update.

rachitnigam, 2019-03-22:
@tedbauer says he got it working.

closed 2019-03-22 by rachitnigam