cucapra/dahlia #164: Our wording for specific notions

sa2257, 2019-04-03:
This issue is created to develop, keep track and improve terminology we come up with to explain specific notions without a corresponding term. (I hope this is not captured elsewhere)

Some examples of these,
Adrian coined the terms
- computational FPGAs
- hardware safety
- low level accelerator design language

Mark Horowitz(?) suggested,
- unsurprising hardware
which was polished by us to
- predictable hardware

rachitnigam, 2019-04-03:
This should go in the paper. If you come up with a new term and a definition, add it in the paper. 

Any reason to use a tracker instead of the paper?

sa2257, 2019-04-03:
In writing an abstract for TECHCON 2019

> (Seashell) permits agility through ensuring generation of predictable hardware with static type checking. 
was suggested to be 

> (Seashell) permits agility by guaranteeing that type checkable code can be synthesized.

> This should go in the paper. If you come up with a new term and a definition, add it in the paper.
> 
> Any reason to use a tracker instead of the paper?

True,
the idea was to make it easily accessible and keep it as a glossary.

rachitnigam, 2019-04-03:
I propose keeping that in the paper itself. Syncing two sources will get increasingly difficult with larger changes. Plus it is a good forcing function for working on the paper itself.

sa2257, 2019-04-03:
In writing an abstract for TECHCON 2019

> (Seashell) permits agility through ensuring generation of predictable hardware with static type checking. 
was suggested to be 

> (Seashell) permits agility by guaranteeing that type checkable code can be synthesized with static type checking.

which doesn't differentiate from HLS tools (which is the intention)

a compromise is suggested with 

> (Seashell) permits agility by guaranteeing that type checkable code generates predictable hardware with static type checking.

To which @rachitnigam pointed out 

> Type checking and precision has nothing to do with each other. You can have a different untyped DSL like Spatial that also generates predictable code because they have their own RTL backend.

To unpack the issue, initial statement is,

1. suggesting static type checking allows predictable hardware generation
2. predictable hardware generation with static type checking allows agility (productivity, if you will)

I think the final suggestion clears up 1. _"type checkable code generates predictable hardware"_
2. is a little far-fetched. The simplistic idea I try to convey is _static type checking is faster than synthesis_, therefore _it's more agile_. But if synthesis is the intention, then a simple C program is more easily synthesized with HLS than our language, so it was important to qualify the design scope to predictable hardware.

sa2257, 2019-04-03:
> I propose keeping that in the paper itself. Syncing two sources will get increasingly difficult with larger changes. Plus it is a good forcing function for working on the paper itself.

fair enough. I don't yet have an efficient way to use the sources for the paper. I'll move this over there once I have it.

rachitnigam, 2019-07-24:
Clarification for predictable hardware added to the paper.

closed 2019-07-24 by rachitnigam