cucapra/dahlia #165: Buildbot seems to disregard estimate flag in make configuration

sa2257, 2019-04-03:
This is a hunch not fully verified, but Buildbot seems to ignore estimate flag in [this](http://gorgonzola.cs.cornell.edu:8000/jobs/en0qLWv8k9k/log.txt)

sampsyo, 2019-04-03:
In that log, Buildbot seems to be doing the right thing. See this line:

```
$ /seashell/buildbot/instance/onxilinx.sh make SDSFLAGS=-perf-est-hw-only
```

So the problem is probably in the Makefile. Maybe it needs to use `SDSFLAGS ?= ...` instead of `SDSFLAGS := ...` to allow this variable to be overridden.

sa2257, 2019-04-03:
Yes but the command doesn't have the flag.
`sds++ -sds-pf zc706  -sds-hw viterbi viterbi.cpp -sds-end -O3 -Wall -Wno-unused-label  -clkid 3 -poll-mode 1 -verbose -I../../common local_support.o support.o viterbi.o main.o -o viterbi`

sa2257, 2019-04-03:
Oh I see, it's possible Rachit's makefile uses `SDSFLAGS` only for the passed in flags. I submitted a test a while back, but it's blocked by full synthesis runs. I'll report ones it runs.

sampsyo, 2019-04-04:
Indeed—but the `make` command is getting invoked correctly, which is Buildbot's job. So the problem must be in how the Makefile uses that variable (not in Buildbot itself).

sa2257, 2019-04-04:
I see what you mean. 

It seems Buildbot relies on having `SDSFLAGS` in the makefile. 

I don't know which strategy is better,
- assuming the makefiles will always have `SDSFLAGS` or
- discarding estimate flags (and others) for makefile runs and use the makefile configurations only.

I like the latter personally, just because it allows any makefile to be used with Buildbot. My take is, Makefile runs are not the common case, but what we can use for specific cases where Buildbot doesn't support or until we get Buildbot support configurations. In that light, it seems prudent to allow any Makefile run, and not get disrupted by other flags. 

However, I cannot think of a way how to use ESTIMATE variable in this light, as the makefile has to reflect that before it gets passed in. One idea is to use batch scripts for this. We can generate the ads command in the makefile from the batch script and compose it before submitting to Buildbot.

sa2257, 2019-04-04:
relevant code in `worker.py`
```
    task.run(
        prefix + ['make', 'SDSFLAGS={}'.format(sdsflags)],
        timeout=config["SYNTHESIS_TIMEOUT"],   
        cwd=CODE_DIR, 
```

sampsyo, 2019-04-04:
I think the most straightforward way to address this is for the Makefile to be self-sufficient in the sense that `SDSFLAGS` is defined to do the sensible thing, including the platform and all that. But the buildbot can *also* provide `SDSFLAGS`, completely overriding the built-in definition. Nice and clean!

sampsyo, 2019-04-06:
This is resolved in #163.

closed 2019-04-06 by sampsyo