cucapra/dahlia #166: Buildbot is agnostic to the kernel name

sa2257, 2019-04-03:
Our simplistic way of running Buildbot included the assumption that the bitstream will be named `sdsoc`. This is not true for the new makefiles, which uses `KERNEL`. I suppose, to make it agnostic of how the bitstreams are generated, we can force the makefiles to always call it `sdsoc`?

rachitnigam, 2019-04-03:
Can you link to the code that uses this assumption?

sampsyo, 2019-04-04:
That would be this string constant:
https://github.com/cucapra/seashell/blob/23bce0220d2442748621bf4260a16fba5dd5e7ec/buildbot/buildbot/worker.py#L17

And its use in this command, which runs the executable on the ZedBoard:
https://github.com/cucapra/seashell/blob/23bce0220d2442748621bf4260a16fba5dd5e7ec/buildbot/buildbot/worker.py#L363

(It's also used in the non-Makefile codepath in an `-o` argument to the compiler. But the Makefile path doesn't know about it.)

sa2257, 2019-04-04:
Couple of thoughts,
- How about giving the bitstream an extension such as `.bit` and let Buildbot pick the extension? Forcing either the name `sdsoc` or `(KERNEL)` seems to be restricting Buildbot. (Also I presume there's no easy way to detect the bitstream otherwise?)

sampsyo, 2019-04-04:
A simple way to do this would be to require the Makefile to use a second variable, called `TARGET`. The Makefile would declare something like this:

```
TARGET ?= $(KERNEL)
```

Then the Buildbot could pass in something like `make TARGET=exe` to be certain that the executable is called `exe`.

sampsyo, 2019-04-10:
It seems pretty straightforward to me to clean up old data from the ZedBoard. In fact, when we ship other files (i.e., data files!) along with executables, we'll want to make sure those don't build up too much too.

I say let's do the thing I proposed above! Buildbot can just pass `make TARGET=exe` or whatever and then know that the build product is supposed to be called `exe`. Want me to try implementing that?

I don't quite understand this caveat:

> we won't be able to use a generic makefile to test something quickly

It still seems pretty easy to write a Makefile for a new app. It just has to include a `TARGET ?= ...` line and make sure it builds `$(TARGET)` in its link step. Our generic Makefile snippet can do that.

sa2257, 2019-04-11:
> It seems pretty straightforward to me to clean up old data from the ZedBoard. In fact, when we ship other files (i.e., data files!) along with executables, we'll want to make sure those don't build [...]

Okay. Shall we try that then? 
Change from the makefile side is just adding `.exe` to the executable. I can add that now.

> I don't quite understand this caveat:
>
>
> It still seems pretty easy to write a Makefile for a new app. It just has to include a `TARGET ?= ...` line and make sure it builds `$(TARGET)` in its link [...]

That's to say, SDSoC doesn't use a distinct extension for their bitstreams for us to automate running on the board for any benchmark or application we come across (provided they use Xilinx tools and provide configuration for the board) without changing their makefile. But I don't see an immediate solution; long-term, our normal buildbot flow can take care of these since it has more control.

sampsyo, 2019-04-11:
Happy to try it if you want!

> Change from the makefile side is just adding `.exe` to the executable. I can add that now.

Can you remind me why we the executable needs a suffix? I was proposing that the executable _just_ be called `exe` or `sdsoc` or whatever (when running in Buildbot). The Makefile change I was imagining was this one I mentioned above: https://github.com/cucapra/seashell/issues/166#issuecomment-479954823

> That's to say, SDSoC doesn't use a distinct extension for their bitstreams for us to automate running on the board for any benchmark or application we come across

Hmm, I think I'm misunderstanding something about SDSoC. I thought that—for the ZedBoard at least—SDSoC generated an `sd_card` directory, and all we needed to do is copy _everything_ in there to the board. Is that incorrect? If not, can you explain more about why we need a "distinct extension"?

sa2257, 2019-04-11:
> Can you remind me why we the executable needs a suffix? I was proposing that the executable _just_ be called `exe` or `sdsoc` or whatever (when running in Buildbot). The Makefile change I was imagining was this one I mentioned above: [#166 (comment)](https://github.com/cucapra/seashell/issues/166#issuecomment-479954823)
> 

Ah! `exe` threw me off. Shall we do `sdsoc`? I don't think we need to change Buildbot at all then. I mistakenly imagined [#166 (comment)](https://github.com/cucapra/seashell/issues/166#issuecomment-479954823) attempts to use `KERNEL`. We can add the TARGET to Makefile. But makefile can just produce `sdsoc`.

sampsyo, 2019-04-11:
I don't think the actual name of the executable matters. It could even be a configuration option. :smiley:

I do intend to change Buldbot to pass `TARGET=some_name_here` to the Makefile knows what executable to build. When this is ommitted, like when you just type `make` at the command line, `TARGET` will be `$(KERNEL)`.

rachitnigam, 2019-04-11:
How do we specify the `TARGET` in the CURL options? Going back to the conversation in #163, it would somewhat simpler if the interface for makefiles used by buildbot is kept simple to avoid issues like options or `TARGET` being ignored.

sampsyo, 2019-04-11:
In what I was proposing, I would *not* specify `TARGET` a job parameter. It would be fixed as `sdsoc` or `exe` or whatever, and provided by Buildbot on the command line to _all_ jobs.

Makefiles in this scenario have two options: use `TARGET ?= $(KERNEL)` or something to get a sensible name when running locally and a fixed name when running on Buildbot, or just hard-code our executable name. In the latter case, there's no chance of an option being ignored.

sampsyo, 2019-04-11:
Fixed in #176.

closed 2019-04-11 by sampsyo