cucapra/dahlia #167: Buildbot doesn't show job name during make stage

sa2257, 2019-04-04:
Buildbot job page doesn't indicate which job it's running. The last bit of logs is also not indicating this. It'll be nice to see which job it is, without digging into files.

sampsyo, 2019-04-04:
It sounds like you'd like for the job metadata to include some sort of string configuration that can be construed as the job's "benchmark name." Can you expand a little bit about how you would prefer (as a user) to specify this benchmark name? For example, you might want to derive it from the directory name that was upload. Or you might want to specify it explicitly. There are many possible options, but thinking through what would be both usable and implementable would be super useful.

sa2257, 2019-04-09:
Ideally, the information to keep track of are,

1. benchmark name
2. if exists, algorithm name
3. version name
4. if our language is used or not
5. if estimate or not
6. if with directives or not
7. what platform is used
8. which clock is used
9. job failed or not

We already have 9. 
Normal flow (without makefiles) also keep track of 1, 4 and 5. 
Make flow can keep track of 5 if input makefiles are in a certain format and follows some conventions. I assume it to handle 4 as before. 

Two solutions we seem to have are,
- use the fact that some of the information is in the fuse-benchmarks folder structure to send that over to buildbot
- use some sort of configuration file to send the information over to buildbot as @rachitnigam suggested. We can use a default configuration otherwise. (1- hw file name, assume one file, 2- NA, 3- baseline, 4- yes, 5- yes, 6- yes, 7- zed, 8- id 3)

sampsyo, 2019-04-10:
Hi! I actually don't think it's a foregone conclusion that the buildbot itself has to keep track of _all_ of that metadata about a job. I agree it's really nice to have at least some of it so that it's easier to tell jobs apart from one another in the web interface, but for the purposes of running _experiments_, the batching script should keep track of things like the clock, the platform, etc.

@sa2257, did you have a specific use case in mind for buildbot keeping track of a _full_ catalog of metadata about a job?

sampsyo, 2019-04-11:
Sure! All of those things are available by picking through the log. If you're interested in seeing those summarized, please leave your thoughts about how to accomplish that (whenever you want to address it)!

rachitnigam, 2019-07-24:
Looks like this has been fixed. Reopen if not.

closed 2019-07-24 by rachitnigam