cucapra/dahlia #169: Buildbot Areesh access failed

sa2257, 2019-04-06:
Noticed a failure to upload to Buildbot. Will update if it's persisting.

```
2019-04-05T23:17:08.791384 $ sshpass -p root scp -r /seashell/buildbot/instance/jobs/1XViEk31y8I/code/sd_card/BOOT.BIN /seashell/buildbot/instance/jobs/1XViEk31y8I/code/sd_card/README.txt /seashell/buildbot/instance/jobs/1XViEk31y8I/code/sd_card/image.ub /seashell/buildbot/instance/jobs/1XViEk31y8I/code/sd_card/_sds /seashell/buildbot/instance/jobs/1XViEk31y8I/code/sd_card/fft zb1:/mnt
channel 0: open failed: connect failed: No route to host
stdio forwarding failed
ssh_exchange_identification: Connection closed by remote host
lost connection
2019-04-05T23:17:12.318759 command failed (1)
2019-04-05T23:17:12.319075 state changed to failed
```

sampsyo, 2019-04-06:
Looks like the Raspberry Pi is up and running, but it can't reach the ZedBoard. I SSH'd into `en-ec-asa2257-b.ece.cornell.edu` successfully, but I can't `ping 192.168.1.2` from there. Maybe it got disconnected?

sa2257, 2019-04-08:
Nothing looks amiss, I power cycled `zb1`.

sa2257, 2019-04-09:
`zb1` is back online.

closed 2019-04-09 by sa2257