cucapra/dahlia #172: Buildbot error in workerpy at if task.get('platform')

sa2257, 2019-04-08:
```
2019-04-08T15:10:32.494647 state changed to make
2019-04-08T15:10:34.183959 acquired in state makeing
2019-04-08T15:10:34.185438 Traceback (most recent call last):
  File "/seashell/buildbot/buildbot/worker.py", line 126, in work
    yield task
  File "/seashell/buildbot/buildbot/worker.py", line 220, in stage_make
    if task.get('platform'):
AttributeError: 'JobTask' object has no attribute 'get'

2019-04-08T15:10:35.324036 state changed to failed
```

rachitnigam, 2019-04-08:
Job link?

sa2257, 2019-04-08:
http://gorgonzola.cs.cornell.edu:8000/jobs/F1Z5p1JEccg.html

sa2257, 2019-04-11:
Buildbot still fails.
```
2019-04-11T02:43:52.392167 Traceback (most recent call last):
  File "/seashell/buildbot/buildbot/worker.py", line 126, in work
    yield task
  File "/seashell/buildbot/buildbot/worker.py", line 220, in stage_make
    if task.get('platform'):
AttributeError: 'JobTask' object has no attribute 'get'
```

sampsyo, 2019-04-11:
Huh; I did fix this. Maybe I somehow messed up the deployment? Can you perhaps check whether this is the same thing that I already fixed or a different bug?

sa2257, 2019-04-11:
Yup, it's fixed. It just wasn't deployed.

commit 746a259a (2019-04-18): Fix #172

This was my fault, introduced in
9bf5aaa7c848e687a721163f5d5f0c2ddd531c85! I forgot tasks weren't proper
dictionaries.
closed 2019-04-11 by sa2257