cucapra/dahlia #174: Visualization Backend Proposal

sgpthomas, 2019-04-10:
This is a proposal for adding a backend to generate GraphViz files from fuse programs. The very high level idea is to represent data and control flow through a program in a layout that's reminiscent of circuit diagrams. 

Declarations are represented as blocks with name and type.
For array values, we display name, type, and number of banks:
[image]

For loops:
Array access form a sort of mux, array writes are a sort of demux
We can display offsets of the iterator variable on the edge from the control node that represents the iterator.
[image]
corresponds to:
```
decl a: bit<32>[4 bank 4];
decl b: bit<32>[4 bank 4];

for (let i = 0..4) unroll 4{
  b[i] := a[i];
}
```
I don't know about while loops

Parallel sequencing is trivial, you just don't do anything special
Linear sequencing will require some sort of registers between stages

sampsyo, 2019-04-10:
Cool! This is a tiny, superficial thing, but it occurs to me that it might be perfectly legible—and a little more compact—to combine the name and type "blocks" in an array representation. The top block could just say "a : int32" instead of having two blocks, one that says "a" and another for "32."

rachitnigam, 2019-04-10:
Also, something to note about pretty printing:

1. The `toString` method for `Type` is already overloaded so doing something like `s"${pretty(expr)}: $t"` will do the right thing.
2. It would be worth defining a separate `PrettySyntax.scala` file that pretty prints AST nodes for the diagrams to reduce the noise from the big blocks (`EArrAccess(...)` would just become `a[i]`)