cucapra/dahlia #178: Fuse Header Files

tedbauer, 2019-04-19:
Add a way to allow Fuse files to include header file definitions.

sampsyo, 2019-04-20:
I talked with @rachitnigam about this a little bit today—it sounds like the plan is something like this:

- Currently, compiling any Fuse program generates a C++ function implementation called `kernel` that contains the top-level commands in the file. We should change the compiler to _not_ emit this function when there are now top-level commands. Then, a "header file" is not really different from an ordinary Fuse source file—there's no special switch or filename extension anything; it's just a Fuse file that consists only of declarations.
- The "import" statement currently only lets you declare stuff from proper C++ headers (AFAIK). We'll need to extend it to let you also include Fuse files, in which case it should (also) import the _types_ defined there.

The alternative to doing all this is just to duplicate all the relevant type definitions in a C++ header file and in the Fuse code. While that redundancy is annoying, it does have one benefit: this way, the C++ code can be compiled without _first_ compiling the Fuse source code. In the land of "Fuse header files," the C++ code will need to include a C++ header file that doesn't exist and needs to be generated by compiling the Fuse header.

tedbauer, 2019-04-24:
Does this mean that the compiler will still emit this `kernel` function in the Fuse "implementation" (as opposed to in the "header/signature")?

sampsyo, 2019-04-24:
Yep! We'll need to emit a `kernel` function implementation from any file that contains commands.

rachitnigam, 2019-04-26:
Instead of doing complicated multi file compilation, we have instead decided on writing a simple new backend that will just produce C header files.

rachitnigam, 2019-04-30:
Fixed in #186.

sa2257, 2019-05-02:
A brief note from debugging session @tedbauer and I did for the rewrites.
For MachSuite benchmarks, most of fuse rewrites work fine in the existing framework (one with symlinks to baseline local support and headers) as long as the function signatures match. (Note fuse generated cpp is not including any header file). 

For some examples, we added the struct definitions in the fuse file which got converted into the cpp file. And things still work fine. 

It's not clear to me whether we need fuse header files. It's a little hacky that fuse cpp compilation is separate from the local support compilation which uses the signature from header file, which makes the designer make sure these are identical. But other than that, do we run into any critical issues? We can add SDS flags used in baseline header through the symlinks.

rachitnigam, 2019-05-06:
@sa2257 any reason to keep this open? If not, please close it.

sa2257, 2019-05-07:
@rachitnigam I'd like to write down our reasoning with header files. As I noted, it's not clear when we need to use fuse headers. I think the implementation to add headers i8n #186 is clear, but it might be useful justify having them and write it down for future reference.

rachitnigam, 2019-05-07:
Seems like you want a doc on the headers. I can add that.

commit c025be1d (2019-05-07): add documentation for headers. Fixes #178
closed 2019-05-07 by rachitnigam