cucapra/dahlia #193: Views and underlying array being used together

rachitnigam, 2019-05-07:
More views complications. Consider this program:

```
view v_a = a[i! :];
---
v_a[10]; a[20];
```

Fuse will happily mark usage of the second access to `a` as correct because it considers `v_a` to be "just an array". Looks like view creation needs some more complicated logic for safety...

rachitnigam, 2019-05-07:
This came up in the PolyBench example:
```C++
// Sigh, a suffix view doesn't work here...
for (j = i; j < _PB_N; j++) {
  w = A[i][j];
  for (k = 0; k < i; k++) {
    // Because one of these will be a view and 
    // another one will be the underlying array.
    w -= A[i][k] * A[k][j];
  }
  A[i][j] = w;
}
```

It is possible to write a program where we are using a suffix view. Unfortunately, that requires us to reorder the `k` and `j` loops and move the `A[i][k]` access out from the inner loop.

sampsyo, 2019-05-07:
Indeed. I think we need the type checker to "remember" that the parts of `a` that were consumed at view creation _stay_ consumed until `v_a` goes out of scope. It's not too different from a really simple version of "borrowing" in Rust—as long as `v_a` exists, it has "borrowed" part of `a`.

rachitnigam, 2019-05-07:
oh no, not a borrow checker.... /s

In the example, it seems like we want a different behavior entirely: `v_a` and `a` can be used in separate pipeline stages, but never together. If we create the restriction that `a` cannot be used while `v_a` is in scope, we do get safety, but it seems like it would produce very convoluted code.

sampsyo, 2019-05-07:
Yeah, this example is hard. The idea that you'd want to use a view and the original array (or, more generally, two views on the same array) _in different pipeline stages_ is really not what I was imagining until seeing this example—I would have hoped that the mutual exclusion afforded by view consumption would be more "lexically scoped" than that. Seems like we need to think about this more…

rachitnigam, 2019-05-07:
The never ending nightmare of views continues...

rachitnigam, 2019-05-07:
Some thoughts about what views mean/how to resolve this issue:

[Gadgets_ Unified theory of memory accesses.pdf](https://github.com/cucapra/seashell/files/3154102/Gadgets_.Unified.theory.of.memory.accesses.pdf)

PR #196 (2019-05-20): Gadget checker
Fixes #193
closed 2019-05-20 by rachitnigam