cucapra/dahlia #197: Buildbot changes needed to run MachSuite executables on zb1

sa2257, 2019-05-29:
Currently Buildbot runs executables on zb1 with `ssh zb1 /mnt/kernel`.

MachSuite apps have presented two problems with this approach. 
1. data files need to be copied (they're copied over to buildbot at job submission)
2. need to log into zb1 and run executable from /mnt because file paths in the executable are relative

It'll be nice to maintain buildbot capability to run jobs without files and with generic names such as `sdsoc`.

sampsyo, 2019-05-29:
Hi! I agree with item 2—that's an easy fix we can do right now (just `cd` into the right directory before running). But I don't understand item 1—what's the alternative to copying data files? What does it mean to "maintain buildbot capability to run jobs without files"? If the benchmark doesn't need to ready any files, it doesn't have to… but if it needs input files, I don't see any way around copying them around.

sampsyo, 2019-05-29:
#199 is that aforementioned quick fix, for your reviewing convenience.

sa2257, 2019-05-30:
> But I don't understand item 1—what's the alternative to copying data files? What does it mean to "maintain buildbot capability to run jobs without files"? If the benchmark doesn't need to ready any [...]

Yup, I just meant the change should work when it doesn't need input files. ie. #200 fails in that regard.

sa2257, 2019-06-03:
#199 allows makefiles to run with data files. #200 allows regular flow to run with data files.

closed 2019-06-03 by sa2257