cucapra/dahlia #202: Bug in declaring multidimensional arrays

KarlMobed, 2019-06-18:
I have run into a problem with declaring arrays with the let binder.

`decl force : float[3][3]; force[1][1] := 2.0;`

This code runs fine; however, replacing decl with `let` throws an error saying that the element (`force`) is not bound in the scope.  

I have replicated this problem with single-dimensional and a variety of multidimensional arrays of all types, including records.

rachitnigam, 2019-06-18:
See the type checker tests for array liberals in https://github.com/cucapra/seashell/blob/master/src/test/scala/TypeCheckerSpec.scala#l1124

Starting line 1124

rachitnigam, 2019-06-18:
I was able to reproduce. Seems like a bug.

rachitnigam, 2019-06-18:
Fixed on master. `git pull && sbt assembly` to rebuild fuse binary.

Thanks for the bug report!

closed 2019-06-18 by rachitnigam

sa2257, 2019-06-20:
@rachitnigam can you add the commit number where this is fixed? 
`let breaks: float[4]` still seem to behave the same.

rachitnigam, 2019-06-20:
See #203 for more details. The current workaround is doing something like:
```
let foo: float[4];
{
 ... code that uses foo
}
... code that doesn't use foo
```