cucapra/dahlia #205: Arrays translated incorrectly

ViviYe, 2019-06-21:
When compiling a .fuse file,
` let temp: bit<32>[2048]; `
was translated into
` bit<32> temp`
when the expected result should be ap_int<32>[2048].

This still happens for fuse programs that only has one line, as above. 
 'bit' did not correctly turn into 'int' and the '[]' went missing.
The problem might relate to the this issue [https://github.com/cucapra/seashell/issues/202](url)

This only seems to happen for arrays. let temp: bit<32>; can be correctly translated into ap_int<32> temp

sampsyo, 2019-06-21:
That does look like a bug! Perhaps this line needs a recursive call to `emitType`?
https://github.com/cucapra/seashell/blob/79b297306035a24d3af9913cd34f8ad583a529bb/src/main/scala/backends/VivadoBackend.scala#L54

rachitnigam, 2019-06-21:
Why is there no facepalm 🤦‍♀️ emoji in GitHub.

Will fix. Thanks for the bug report @ViviYe!

commit d1d7df64 (2019-06-21): fixes #205
closed 2019-06-21 by rachitnigam