cucapra/dahlia #206: Tests failing out of the box

gussmith23, 2019-06-25:
Trying to run tests, but getting errors out of the box:
```
[info] RunTests:                                                                                                                                                                         
[info] - src/test/should-run/matadd.fuse *** FAILED *** (45 milliseconds)                                                                                                                
[info]   fuselang.CompilerError$HeaderMissing: Header json.hpp is missing from location src                                                                                              
/main/resources/headers.                                                                                                                                                                 
[info]   at fuselang.GenerateExec$.$anonfun$generateExec$1(GenerateExec.scala:55)                                                                                                        
[info]   at fuselang.GenerateExec$.$anonfun$generateExec$1$adapted(GenerateExec.scala:53)                                                                                                
[info]   at scala.collection.immutable.List.foreach(List.scala:392)                                                                                                                      
[info]   at fuselang.GenerateExec$.generateExec(GenerateExec.scala:53)                                                                                                                   
[info]   at fuselang.Main$.$anonfun$runWithConfig$4(Main.scala:79)                                                                                                                       
[info]   at scala.util.Either.flatMap(Either.scala:341)                                                                                                                                  
[info]   at fuselang.Main$.runWithConfig(Main.scala:77)                                                                                                                                  
[info]   at fuselang.AsyncRun$.$anonfun$compileAndEval$1(RunTests.scala:38)                                                                                                              
[info]   at scala.concurrent.Future$.$anonfun$apply$1(Future.scala:658)                                                                                                                  
[info]   at scala.util.Success.$anonfun$map$1(Try.scala:255)                                                                                                                             
[info]   ...                                                                                                                                                                             
```

reproduce:
```
➜  ~ git clone https://github.com/cucapra/seashell tmp                                                                                                                                   
Cloning into 'tmp'...                                                                                                                                                                    
remote: Enumerating objects: 70, done.                                                                                                                                                   
remote: Counting objects: 100% (70/70), done.                                                                                                                                            
remote: Compressing objects: 100% (53/53), done.                                                                                                                                         
remote: Total 8223 (delta 33), reused 40 (delta 17), pack-reused 8153                                                                                                                    
Receiving objects: 100% (8223/8223), 3.20 MiB | 6.29 MiB/s, done.                                                                                                                        
Resolving deltas: 100% (5302/5302), done.                                                                                                                                                
➜  ~ cd tmp                                                                                                                                                                              
➜  tmp git:(master) sbt compile                                                                                                                                                          
[info] Loading settings for project tmp-build from assembly.sbt ...                                                                                                                      
[info] Loading project definition from /home/gussmith/tmp/project                                                                                                                        
[info] Loading settings for project tmp from build.sbt ...                                                                                                                               
[info] Set current project to Fuse (in build file:/home/gussmith/tmp/)                                                                                                                   
[info] Executing in batch mode. For better performance use sbt's shell                                                                                                                   
[info] Updating ...                                                                                                                                                                      
[info] Done updating.                                                                                                                                                                    
[info] Compiling 27 Scala sources to /home/gussmith/tmp/target/scala-2.12/classes ...                                                                                                    
[info] Done compiling.                                                                                                                                                                   
WARNING: An illegal reflective access operation has occurred                                                                                                                             
WARNING: Illegal reflective access by com.google.protobuf.UnsafeUtil (file:/home/gussmith/.                                                                                              
sbt/boot/scala-2.12.7/org.scala-sbt/sbt/1.2.8/protobuf-java-3.3.1.jar) to field java.nio.Bu                                                                                              
ffer.address                                                                                                                                                                             
WARNING: Please consider reporting this to the maintainers of com.google.protobuf.UnsafeUti                                                                                              
l                                                                                                                                                                                        
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access                                                                                               
operations                                                                                                                                                                               
WARNING: All illegal access operations will be denied in a future release                                                                                                                
[success] Total time: 9 s, completed Jun 24, 2019, 6:16:25 PM                                                                                                                            
➜  tmp git:(master) sbt test                                                                                                                                                             
[... cut ...]
[info]   at scala.collection.immutable.List.foreach(List.scala:392)                                                                                                                      
[info]   at fuselang.GenerateExec$.generateExec(GenerateExec.scala:53)                                                                                                                   
                                                                                                                                                                   
etc...
```

Am I missing something dumb here?

rachitnigam, 2019-06-25:
Can you try running: `sbt "; clean; getHeaders; test"`?

The `fuse run` command used to generate c++ executable requires a json header file in `src/main/resources/headers/`. The link to the header file can be found in `build.sbt` if `getHeaders` doesn't work for you.

rachitnigam, 2019-06-25:
Found the bug. Fixed by 06d7a8c3a9c7b2328a03853219f69555180f291f.

closed 2019-06-25 by rachitnigam