cucapra/dahlia #211: Tracker: Benchmarking codegen hacks

sampsyo, 2019-07-08:
This issue is about things to change in code generation which are **just hacks** to make benchmarking easier. The idea is to avoid manual modifications to the output C++ code so we can directly evaluate what we generate. The spirit behind these hacks is *not* elegance; we just need to get the job done by any means necessary in the easiest possible way.

Here's what I am aware of so far:

- [ ] *Software integers.* It's useful for debugging purposes to compile the "HLS" output code using a normal software compiler (i.e., gcc or clang). This means that the code needs to avoid using `ap_int` when doing that. I can see several ways to do that:
    - Insert an `#ifdef` (which activates only in an HLS compiler) around a `template` hack to replace `ap_int` with `int`.
    - Add a new backend mode, called like `vivado-sw` or something, that just generates `int` in place of `ap_int`.
    - Like the above, but use [`ac_int`](https://hlslibs.org/#collapseACDatatypes), as mentioned in https://github.com/cucapra/seashell/issues/105#issuecomment-475416029. (This would be nice but not in the spirit of a dumb-as-possible hack.)
- [ ] *SDSoC pragmas.* We need to add the `#pragma SDS` nonsense that makes the whole SDSoC compiler toolchain go. An easy, inelegant way to do this would be to add a construct to the language that just specifies a string to output before the function declaration; we pass this string through unchanged.
- [ ] *Trip count pragmas for estimation.* In estimation mode (but not when generating real hardware), the HLS tool wants to know the estimated trip count for `while` loops (which it obviously can't deduce by itself) so it can estimate the time the execution will take. We could also use the pass-through string approach for this.

sa2257, 2019-07-09:
For *Software integers* I'm adding 
```
#ifdef __SDSCC__
#include "ap_int.h"
#else
template < int N >
using ap_int = int;
#endif
```

sampsyo, 2019-07-15:
Moved character literals to #105. I'm closing this issue, since we've done the big ones—there are still outstanding differences in the manually-hacked C++, but we can open individual issues for those after we understand them.

closed 2019-07-15 by sampsyo