cucapra/dahlia #217: Unsolved HLS Problems for Intel Backend

KarlMobed, 2019-07-15:
This is a list of unsolved problems for future Edan (ha ha, that sucker!) to deal with relating to HLS and the new backend for the Intel fpga.

### 1. Banking in multi-dimensional arrays
     Potential conflict with implementation for banked single array
### 2. Views
     Literally everything

sampsyo, 2019-07-15:
Thanks, @KarlMobed!

To expand on each of these, briefly:

1. The `__attribute__` thingy that the Intel OpenCL compiler uses doesn't seem to be per-bank. Instead, you use a "chunk size" parameter. Maybe we can carefully engineer the chunk size to emulate multi-dimensional banking.
2. Our C/C++ implementation of views uses pointer arithmetic, which isn't available for OpenCL arrays. 🤷‍♀

rachitnigam, 2020-08-12:
Stale issue. @KarlMobed never implemented an MVP for the backend that could be merged with master.

closed 2020-08-12 by rachitnigam