cucapra/dahlia #218: Experiment: C level array parititoning vs partition pragma

rachitnigam, 2019-07-15:
## Experiment

Hypothesis: Manually partitioned arrays at C level perform as well as the partition pragma

Program 1: Two dimensional array with partition pragmas on both dimensions (say with 2 banks). Access each element of the array and increment it by 1 for 1000 iterations.

Program 2: 4 two dimensional arrays representing the partitions for the array in prog 1. A function with manual arbitration logic for selecting the right array based on the input arrays. Only use the function to access and update the elements of the arrays in the same way that prog 1 does.

## Rationale
I'm writing the Views section with a "Challenges in implementation" subsection which shows why the primitives exposed by various HLS backends are not the right one. 

One challenge in implementing views is we can't take slices of multidimensional arrays in C. We assume that this will be easy to do in a hypothetical RTL backend. Let's try to implement this at the C level and see how well it does and make sure we have an airtight argument backed up by an experiment.

@sampsyo can I ping you to do this? I'll write up the C code tonight if someone can try to push it through synthesis / estimation tomorrow.

sampsyo, 2019-07-16:
Good call! I would be interested to see this too. I really have no idea whether the tool’s built-in banking will have “magical” efficiency we can’t attain otherwise. Seems like we could even test this with a single-dimensional array, although a multi-dimensional one is of course where it really matters for our backends.

Happy to run some jobs when I get a moment. But FWIW, here is my favorite way to run jobs from home:

    $ ssh -NL8000:gorgonzola:8000 waffle
    $ export BUILDBOT=http://localhost:8000

rachitnigam, 2019-07-24:
Tracked in https://github.com/cucapra/fuse-benchmarks/pull/93

closed 2019-07-24 by rachitnigam