cucapra/dahlia #221: Handling negative numbers

sa2257, 2019-07-18:
I'm getting errors when using negative numbers. 
```
let a = -1;
``` 
gives me
```
[Parsing error] [15.22] failure: '(' expected but '-' found
```

with brackets it gives me
```
[Parsing error] [15.23] failure: Base Failure
```

when used like this
```
let a = (0-1);
``` 
it generates cpp, but with one bit
```
ap_int<1> MISMATCH_SCORE = (0 - 1);
```

https://github.com/cucapra/fuse-benchmarks/issues/66

sampsyo, 2019-07-18:
Curiously, parenthesizing the negative yields a less scrutable error message:

```
[Parsing error] [1.10] failure: Base Failure

let a = (-1);
         ^
```

Anyway, as for your current workaround, how about giving those explicit types?

```
let MISMATCH_SCORE: bit<2> = (0-1);
let GAP_SCORE: bit<2>      = (0-1);
```

sa2257, 2019-07-18:
> Curiously, parenthesizing the negative yields a less scrutable error message:
>
> ```
> [Parsing error] [1.10] failure: Base Failure
>
> let a = (-1);
>          ^
> ```
>
> Anyway, as for your current workaround, [...]

Oh yes! :man_facepalming: That's way better than manually changing the cpp.

commit f1961e23 (2019-07-21): Fixes #221. Parse negative integers
closed 2019-07-21 by rachitnigam